rstrouse/ESPSomfy-RTS #279: problem after power outage

senolsun, 2024-02-16:
### Hardware

ESP32

### Firmware version

v2.3.3

### Application version

v2.3.3

### What happened? What did you expect to happen?

After a power outage, my esp somfy does not boot up properly. it shows a blueish purplish light on it and does not respond to requests until i remove the power from and and power it back on. usually it takes some time for my pfsense router to boot up and i thought it had something to do with the DHCP not providing the IP address so I set my espsomfy to use a static IP but the issue remains. This does not happen all the time but I'd say 1/3 times this issue is there. 

### How to reproduce it (step by step)

```bash
cut power to your house and turn it back on
```

### Logs

```bash
no logs
```

rstrouse, 2024-02-16:
After the first attempt to connect it will try the network.   If it cannot find the network it will fall back to the AP.   After 3 minutes if the AP is not connected to it will then again attempt to connect to the network and if it is not yet up it will cycle again.

senolsun, 2024-02-16:
last time this happened I had to manually unplug it and replug it to get it going, it did not do it automatically as you mentioned.

rstrouse, 2024-02-16:
Which ESP32 do you have that has a blue light?

rstrouse, 2024-03-10:
@senolsun install the v2.4.1 pre-release.  This may have to do with the router startup and the availability of access points.   This version now listens for better access points as they come online and will switch dynamically to the best connection.

closed 2024-03-23 by rstrouse