rstrouse/ESPSomfy-RTS #280: Implementing additional commands from Somfy RS485 RTS transmitter

perfrankling, 2024-02-18:
### Describe you new feature you'd like

In addition to a handheld Somfy remote, I have a Somfy RS485 transmitter. The transmitter can send somewhat different RTS commands than the handheld remote.
* It has separate "Stop" and "Favorite" commands. The Stop command does nothing if the shade is not moving, and always stops the shade if it is moving. The favorite command always moves the shade to the favorite position, even if the shade is already moving. The behavior is thereby not the same as when sending the "My" command from a handheld remote.
* The "Tilt+" and ""Tilt-" commands can be modified to adjust the tilt amplitude. The tilt amplitude can vary between 1 and 127

This is the device: https://service.somfy.com/downloads/br_v5/rs485rts_transmitter_manual.pdf

For integration with eg wall buttons, it would be quite beneficial to have these commands available to send from ESPSomfy-RTS.

This is a log of sending first a "Favorite" and then a "Stop" command:

`[{"encKey": 137,
  "address": 13486605,
  "rcode": 5773,
  "command": "My",
  "rssi": -22,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [2253,2495,2514,2495,2515,2499,2511,2499,2511,2497,2512,2474,2536,2476,2508,2501,2508,2496,2509,2497,2534,2471,2535,2471,4836,1334,1318,607,639,640,649,1262,1303,622,658,1267,671,621,1278,641,646,1269,1287,645,644,630,661,1266,693,586,1357,594,618,649,635,640,654,1269,654,614,658,647,1268,660,640,644,670,613,636,1267,1294,1259,1304,1272,665,625,1290,634,663,627,645,1270,635,628,664,623,1307,651,619,647,634,639,655,1271,652,613,1313,1266,1285,648,642,651,611,640,678,618,647,634,635,641,646,644,628,1288,650,619,1300,641,647,643,628,1263,1302,648,634,635,639,643,650,1260,1364,1211,660,624,1291,636,635,1292,1293,639,633,1275,647,662,632,615,671,622,652,614],
  "time": "2024-02-18T16:36:04.155+0100"
},
{"encKey": 137,
  "address": 13486605,
  "rcode": 5773,
  "command": "My",
  "rssi": -21,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [584,2642,2512,2495,2515,2495,2509,2495,2513,2496,2512,2495,2511,2497,4813,1336,1300,632,639,644,655,1262,1298,627,636,1291,644,625,1307,624,667,1258,1286,644,647,633,636,1266,668,630,1287,654,639,626,662,616,654,1270,653,636,657,620,1293,661,622,644,620,657,635,1268,1297,1283,1275,1275,666,614,1298,629,687,599,648,1269,661,631,639,644,1278,653,641,645,633,636,639,1294,633,634,1290,1274,1288,644,647,659,603,640,646,644,632,663,634,649,618,650,636,1291,1295,633,637,664,620,643,679,1244,1274,648,636,639,654,620,644,1283,1291,1270,660,632,1273,650,635,1285,1291,633,635,1279,668,609,1286,1273,675,614],
  "time": "2024-02-18T16:36:04.294+0100"
},
{"encKey": 137,
  "address": 13486605,
  "rcode": 5773,
  "command": "My",
  "rssi": -20,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [584,2644,2511,2500,2508,2496,2513,2495,2514,2498,2512,2494,2512,2493,4814,1336,1298,635,625,643,674,1238,1303,647,634,1266,672,620,1279,640,646,1270,1287,646,644,630,660,1266,668,628,1284,640,643,621,682,616,632,1274,674,613,661,625,1294,633,635,641,646,643,630,1285,1290,1281,1282,1269,660,630,1291,629,662,634,639,1270,660,631,640,619,1302,654,641,620,657,640,626,1321,634,617,1283,1270,1307,627,643,648,634,636,641,647,643,629,660,632,649,642,627,1287,1276,649,635,690,595,643,648,1261,1291,644,630,663,633,624,643,1276,1291,1269,683,640,1273,620,656,1265,1298,628,660,1266,669,601,1309,1276,647,611],
  "time": "2024-02-18T16:36:04.434+0100"
},
{"encKey": 138,
  "address": 13486605,
  "rcode": 5774,
  "command": "My",
  "rssi": -21,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [2645,2497,2509,2497,2509,2498,2516,2471,2519,2499,2511,2498,2512,2497,2512,2497,2520,2496,2512,2495,2512,2474,2514,2494,4814,1362,1271,640,653,644,646,1260,1305,1271,1312,1244,1280,641,646,1270,1312,670,605,1253,1309,1268,1286,646,643,628,661,631,640,1269,1287,645,693,586,660,632,644,639,629,661,639,1270,1278,1279,1293,1290,636,624,1306,626,640,650,634,1267,721,558,675,614,1286,643,650,634,635,652,640,1275,661,624,1290,1260,1329,609,655,636,633,648,643,628,660,631,640,642,629,661,634,1274,647,660,1260,643,649,634,639,1267,1293,687,597,648,646,625,659,1259,1291,1285,639,619,676,610,660,621,691,611,658,666,620,618,645,637,665,600,1290,1282,1289],
  "time": "2024-02-18T16:36:10.160+0100"
},
{"encKey": 138,
  "address": 13486605,
  "rcode": 5774,
  "command": "My",
  "rssi": -20,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [658,615,666,620,643,633,660,615,664,619,1300,3197,2535,2474,2518,2499,2514,2499,2512,2498,2510,2498,2512,2471,4841,1361,1252,635,664,618,650,1261,1302,1275,1284,1269,1307,639,642,1279,1292,618,651,1262,1303,1326,1250,617,651,637,661,627,643,1273,1289,644,647,633,635,640,672,622,628,637,640,1281,1294,1263,1298,1280,658,617,1294,635,654,620,645,1284,656,619,670,609,1312,620,695,589,638,661,625,1270,656,640,1282,1269,1287,671,620,631,687,600,639,645,656,637,640,625,643,649,635],
[... 112 lines cut ...]
},
{"encKey": 150,
  "address": 13486605,
  "rcode": 5786,
  "command": "Step Down",
  "rssi": -24,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [592,2640,2507,2504,2508,2474,2536,2473,2515,2499,2512,2497,2512,2497,4816,1339,1298,642,625,1268,1334,1246,647,636,1287,644,646,634,636,1267,1294,1289,651,618,1299,641,624,643,649,636,636,1291,667,612,659,665,1243,1275,1286,1270,1307,1251,1295,659,640,643,652,613,660,627,640,648,634,1260,666,624,1311,1248,670,611,1312,1271,652,610,1309,1268,1286,671,623,1327,591,618,670,611,664,625,1290,1281,1290,1274,1286,1295,1261,1279,1295,635,653,644,622,708,574,1293,1293,641,639,623,652,642,633,1289,640,646,1285,1266,653,674,602,640,649,664,629,614,678,607,675,594,1308,1276,645,637,639,654],
  "time": "2024-02-18T16:44:26.582+0100"
},
{"encKey": 151,
  "address": 13486605,
  "rcode": 5787,
  "command": "Step Up",
  "rssi": -21,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [2622,2483,2529,2481,2529,2483,2527,2477,2514,2506,2515,2484,2516,2488,2523,2484,2516,2490,2523,2490,2525,2487,2502,2510,4816,1373,1247,644,627,1289,1275,1275,659,623,668,628,1287,624,669,1256,1290,1269,657,635,1284,1267,1312,620,647,1281,657,619,669,608,1287,1296,652,611,685,593,1342,1228,1290,656,636,640,649,643,626,636,660,640,620,1271,662,640,1274,1277,666,624,1294,1260,690,594,1296,1267,1296,629,662,1267,668,601,658,640,658,619,1294,1315,1244,1280,1290,1273,1287,1271,1307,640,620,645,656,639,633,1274,1297,632,640,644,629,685,615,1270,653,639,655,620,1293,636,633,647,642,629,660,641,626,668,628,1309,1244,1275,661,624,1294],
  "time": "2024-02-18T16:44:42.905+0100"
}]`

rstrouse, 2024-02-18:
I took a look at the pulses you sent and these simply look like standard 80-bit commands.   Bear in mind I have all of the first 56 bits accounted for but the last 3 bytes of the 80-bit protocol have some mystery bits.   At this point the first of these 3 bytes are always 132 or 196 depending on the command series.  Given the start of the bit sequence this probably starts low for a full beat so the lower 2 bits are always going to be 0 followed by a 1.   I have suspicion that only the upper nibble of this byte contains usable data.  A 4 bit sequence gives a pretty consistent indicator that the follow on bits are coming.

For your my command reference... In this instance the remote simply writes 3 extra bytes into the inter-frame silence which ESPSomfy RTS picks up.   In the first instance this represents a common my command that was repeated twice for a standard 80-bit my command.  Those 3 bytes at the end the my command at 80 bits always writes a value of [196,44,159] for the first frame and a repeat is [132,44,155].   The second frame series is interesting in that it outputs [196,47,250] in the interframe silence.  But I suspect the repeat sent after that where the sequence is [0,0.0] is just being flagged to support the 56-bit motors.  This is true since it is in the repeat phase and the preamble is ignored and a listening 56 bit motor will know how to translate that.

[image]

On the tilt functions these are 80-bit step up/down commands.  The 3 bytes in the encoding come out as follows.  Not sure at all what the encoding is on those 3 bytes but the most significant bit of the lower nibble of byte[1] indicate the direction.  So if bit 4 is 1 then we are stepping up.  The bit packing for the amplitude has to be somewhere in the last 2 bytes of the 80-bit protocol if your assertion on the command is correct.

[image]

perfrankling, 2024-02-18:
Looking at the two final bytes, for the tilt, it seems the first four bits is the command (0b0011 for Tilt apparently), the fifth bit is the direction (1 = minus, 0 = plus), bit 6-12 is the seven bits of the amplitude and the final four bits is some kind of checksum.

rstrouse, 2024-02-19:
Run some more tests to see if that actually aligns.   The step up vs down bit (5) is correct but the command byte is actually obfuscated in the first 56 bits of the frame.  While it may seem like it is encoding the command into that I have plenty of 80-bit examples where that is not the case.  In fact your example above is one of the examples.   Bits 6-12 on the last byte makes some sense but given the description of the frames you gave above it doesn't match.

perfrankling, 2024-02-19:
I'll collect more logs tonight or tomorrow. Is there a simple algorithm or tool to translate the pulse information to bits and bytes?

rstrouse, 2024-02-19:
Its relatively simple.  Unzip the archive below into a directory then browse to the html file in a browser.  You should then be able to take your raw log timing and paste it into the field provided.  Then press process and it will decode the timings.

[RTSTimings.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/14336411/RTSTimings.zip)

rstrouse, 2024-02-25:
@perfrankling Were the contents of the zip file helpful?

perfrankling, 2024-02-25:
> @perfrankling Were the contents of the zip file helpful?

Yes, very. I think I have worked out the meaning of the final three bytes by logging the commands from the RS485 transmitter and my Telis 4 handheld remote, but I need to pull it together, hope to find time later today .

perfrankling, 2024-03-02:
Attached is a spreadsheet of logs of commands sent from my handheld Telis 4 remote (address 4463618) and from my RS485 RTS transmitter (13486605). From the Telis 4, I have sent the commands with a short press and with a long press (generating repeats).

My conclusions regarding byte 8-10 are:

Bit 1: Always "1"
Bit 2: "1" if button is not yet released, ie always "0" for the last repeat. The RS485 transmitter generally sends "1" for the first repeat and "0" for the following
Bit 3-6: Repeat sequence number when button is kept pressed on Telia4. Always "1" from RS485
Bit 7-8: Always "0b00"
Bit 9-20: Payload; See logs
 - For "Fav": "0b001011001001"
 - For "Stop": "0b001011111111"
 - For Step Up/Step down:
     - Bit 9-12: "0b0011"
     - Bit 13: "1" for Down; "0" for Up
     - Bit 14-20: Step size 1-127 (or 0-127?)
  - For "My" and some others bit 13-20 contain another repeat counter
Bit 21-24: Checksum: Bitwise XOR of the 5 other nibbles in bytes 8-10

[RTS 80-bit logs.xlsx](https://github.com/rstrouse/ESPSomfy-RTS/files/14467407/RTS.80-bit.logs.xlsx)

rstrouse, 2024-03-02:
Well you have been busy.

So would you agree that line 193 and 203 represent invalid frames which is why the step intensity is incorrect for those lines?

perfrankling, 2024-03-02:
Yes. Those are the only instances where the computed checksum does not match the final nibble.

perfrankling, 2024-03-16:
Implementing these commands would minimise the risk of issues like these: [https://github.com/rstrouse/ESPSomfy-RTS-HA/issues/44](url)

perfrankling, 2024-04-28:
Any further thoughts on this? Would it be complicated to implement?

rstrouse, 2024-04-29:
I have plans to add this to the decode frame methods then we can figure out how to configure it from the ui.

perfrankling, 2024-04-29:
For my use case the important thing is to be able to send the commands from HA.

Velocet, 2024-04-29:
@perfrankling There are some libs on GitHub already implementing this so called U80 protocol. Since the modulation is the same (with some huge margins) this lib should be able to handle this quite easily... buuut i think the maintainer could answer this way better then myself.

rstrouse, 2024-04-29:
All 80 bits as well as the protocol detection are being captured.   I just needed to get some priorities handled with the v2.4.2 release before adding the data storage and translations for the data.  The translations are very straightforward with the above spreadsheet.

rstrouse, 2024-05-03:
I am going through and adding the decoding/encoding for the step up/down commands and size and have a pretty clear picture of what the indications are in the data for these.  However, it is a bit unclear as to the differences with the my command.  I would have suspected that byte index 8 would contain the data for the favorite command.

What I am seeing from your spreadsheet is that the my command when issued as a general stop has either a 0 or 47 in this byte.  I suspect there is something in this byte that indicates whether it should be favorite or stop.   Theoretically, I suppose it could be 3 commands:
1. 47 Stop - Only perform a stop if there is movement
2. 44 Favorite - Move to the my position regardless of whether there is movement.
3. 0 My - Move to the favorite position if at idle but stop if there is movement.

Can I get you to test the theories when the motor is moving and idle for the stop and favorite commands?

perfrankling, 2024-05-04:
Yes, that is correct, in accordance with my description of "bit 9-20" from 2 Mar above.

* For rows 26-28 I sent the "Fav" command from the RS485 transmitter, which always results in the blind moving to the my position. Byte 8 (column M) is then 44.
* For rows 29-31 I sent the "Stop" command from the RS485 transmitter, which always stops the blind. Byte 8 is 47
* In rows 8-20 I held the my button depressed on the Telis 4 remote. Byte 8 is 0
* In rows 23-24 I pressed the my button on the Telis 4 briefly. Bytte 8 is 0. The blind moves stops if it is moving, or moves to the my position if it is not moving.

There are also differences in the most significant nibble of the final byte:
* Fav command (from RS 485): Always "1001", ie hex 9
* Stop command (from RS485): Always "1111", ie hex F
* My command (from Telis 4): Counting up, starting at 1

rstrouse, 2024-05-04:
Thanks,  I am working in the RTSTimings scripts at this point.  When I get done with these I will zip them up again to make sure the data matches the commands.  I have already added the checksum checks to the code.

[image]

rstrouse, 2024-05-07:
@perfrankling I have added the commands for the Step Up/Down and stepSize attribute as well as the stop and favorite commands.  If you install the v2.4.3 pre-release.   I have also included the RTSTimings page to allow you to inspect the decoding.

[RTSTimings.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/15242077/RTSTimings.zip)

When you issue these commands externally they will show up in the logs as follows where the superscript indicates the step size.
[image]

You can also issue these commands from the virtual remote.
[image]

perfrankling, 2024-05-07:
Great! I am travelling and may not get a chance to test this until Sunday.

perfrankling, 2024-05-12:
Evertyhing seems to work as expected! Thanks you!

I have some (probably unconnected) issues with the Home Assistant integration, will report that separately!

rstrouse, 2024-05-12:
Great! I am going to close this issue for now.

closed 2024-05-12 by rstrouse