rstrouse/ESPSomfy-RTS #282: [INFO] Wind sensor

chemelli74, 2024-02-18:
### Describe you new feature you'd like

Cannot avoid to thank you for your fantastic work again. Is super nice !

I have just one more small question (yes I hope is the last one as well as you are ;-) ).

I configured my shade as awning in ESPSomfy RTS web interface.

I noticed that a `binary_sensor` with "mdi:wind-power-outline" icon was created.
But I didn't yet linked my sensor as a remote.

This means it's detected automatically on my specific scenario/installation ?
Or is created by default and to make it working I need to link it ?
If so the checkbox is the one named "**Has Sun Sensor**" even if it's the wind one right ?

Thank you for your work, time and patience.

rstrouse, 2024-02-18:
The sensors for wind are always added when the motor represents an awning.  This is so you can perform this function from weather reports if you like using an automation.  This function is not overridable on the Somfy protocol and vibration sensors can be uses in lieu of an Eolis or similar.  This sensor is created by default.  In fact you can send that sensor command to the directly to any remote linked to the motor motor and it will spring into action.

The sun sensor is a different duck in that its existence is conditional based on a series of frames sent but they are ignored if the sun flag is turned off.   So if you do not have a sun sensor do not check the Has Sun Sensor.   But if you want to simulate one using an automation you can do that as well and you will get the enable/disable functions as if your remotes support it.  It will also command the motors properly.

chemelli74, 2024-02-18:
I have a vibration sensor, and if I got it right I have to link it as standard remote so I can have always the right awning position.
Just pairing and that's all. Right ?

Thank you for the very quick reply.

rstrouse, 2024-02-18:
Yes link it as a remote for all motors affected by it.  You do not pair it as a separate motor.

chemelli74, 2024-02-18:
Thx, tomorrow with the sun light I'll proceed with the linking and report back.

Cheers

rstrouse, 2024-02-19:
> is expected the "0" in the second column ?

Yes because the sensor frame doesn't actually have a rolling code.  Imagine my surprise with that.  The state is actually pumped into one of the nibbles on the rolling code.  You will see it change if the sensor reports a wind condition then change back after it confirms the wind has subsided.  Depending on which one you have you will see some bursts of output but they are all broadcast frames for the wind sensor.  If you watch the log you should see the activity.

chemelli74, 2024-02-19:
Thank you for all the information and details.
Closing as all seems good.

closed 2024-02-19 by chemelli74