rstrouse/ESPSomfy-RTS #284: Backup doesn't contain/restore MQTT settings

xvisor16, 2024-02-23:
### Hardware

ESP32

### Firmware version

v2.3.1

### Application version

v2.3.1

### What happened? What did you expect to happen?

I wanted to update and this failed with a Github error. After that ESPSomfy crashed at trying to access the webinterface. I reinstalled the ESP32 with the latest onboard file (v2.4.0) through ESPHome. Setup Network and restored the last backup. Everything seemed fine except that MQTT wasn't working anymore. Upon checking all settings MQTT was disabled and all the settings missing  

### How to reproduce it (step by step)

```bash
1. Reinstall from scratch through ESPHome.
2. Install the latest onboard bin file (v2.4.0).
3. Setup networking (WiFi).
4. Recover from the latest backup (was from v2.3.1).
```

rstrouse, 2024-02-24:
MQTT settings are not included in the backup file by design.

Please read the reasoning for this in the wiki.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#backup-and-restore

xvisor16, 2024-02-24:
Thnx, I should probably start re-reading the Wiki every update ;) I do understand the reasoning behind not storing usernames, passwords and wifi information in plain text backups. But isn't it an option to at least store the selected options and the MQTT topic's leaving the server and the credentials out? This would be safe enough wouldn't it?

rstrouse, 2024-02-24:
The MQTT topics and the auto configure option do make sense.  I'll add it to the list.

rstrouse, 2024-02-25:
v2.4.0 release backs up and restores the MQTT settings sans username and password.

closed 2024-02-25 by rstrouse