rstrouse/ESPSomfy-RTS #286: Now starting with USB power adaptar

luismalddonado, 2024-02-25:
### Hardware

ESP32

### Firmware version

2.4.0

### Application version

2.4.0

### What happened? What did you expect to happen?

First of all thanks for this amazing software. But I have a problem using an ESP32 ESP-WROOM-32 NodeMCU Dev Kit C with  Chip CP2102:
Application does not start when USB power adaptar is used.  There are other two situation that make the application works well:
  - Computer USB is used to power on and arduino IDE serial monitor is used
  - USB adaptar is used to power on  with CC1101  E07-M1101D-SMA Transceiver unplugued (3v3 pin unplugued during start `process) and plugged once ESP32 is accesible via web interface

But when starting with USB power adaptar and with Transceiver  connected it does not start

### How to reproduce it (step by step)

```bash
1. Plug in ESP32 with power adaptar
2. ping ESP32 and wait
3. Web interface never gets accesible
```

### Logs

```bash
ping to ESP32. I cannot provide Arudino logs, because when using PC USB port, it works well
Pinging 192.168.3.25 with 32 bytes of data:
Request timed out.   <--- ESP32 powered on via USB power adapter
Request timed out.
Request timed out.
Reply from 192.168.3.25: bytes=32 time=9ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=8ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=6ms TTL=255
Reply from 192.168.3.25: bytes=32 time=7ms TTL=255
Reply from 192.168.3.25: bytes=32 time=5ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.3.25: bytes=32 time=8ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.3.10: Destination host unreachable.
Reply from 192.168.3.10: Destination host unreachable.
Reply from 192.168.3.10: Destination host unreachable.
Reply from 192.168.3.10: Destination host unreachable.
Reply from 192.168.3.25: bytes=32 time=1308ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=16ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.3.25: bytes=32 time=8ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=5ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=9ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Request timed out.
Request timed out.
Request timed out.
```

rstrouse, 2024-02-25:
Which pins are you using for the Transceiver?  Those periods of request timed out suggest that the esp32 is rebooting when it tries to initialize the radio.

luismalddonado, 2024-02-25:
The proposed ones:

      Transceiver      ESP32
1	GND	      GND
2	VCC		      3v3
3	GDO0 	      GPIO 13
4	CSN		      GPIO 5
5	SCK		      GPIO 18
6	MOSI	      GPIO 23
7	MISO	      GPIO 19
8	GDO2 	      GPIO 12

As I mentioned if a start with the Transceiver  without power and power it later everything works.

rstrouse, 2024-02-25:
Lets try this.   Get the web interface up then uncheck the enable radio checkbox and save the transceiver.  Then hook everything up and lets see if it boots off your usb power adapter.  The pins above should not have any boot dependencies.

luismalddonado, 2024-02-25:
> Lets try this. Get the web interface up then uncheck the enable radio checkbox and save the transceiver. Then hook everything up and lets see if it boots off your usb power adapter. The pins above should not have any boot dependencies.

Same issue with uncheck radio checkbox:

C:\Users\luism>ping -t 192.168.3.25
Pinging 192.168.3.25 with 32 bytes of data:
Request timed out.
Reply from 192.168.3.25: bytes=32 time=7ms TTL=255
Reply from 192.168.3.25: bytes=32 time=9ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Reply from 192.168.3.25: bytes=32 time=9ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.3.25: bytes=32 time=7ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=23ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=8ms TTL=255
Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.3.25: bytes=32 time=5ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
Reply from 192.168.3.25: bytes=32 time=5ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Request timed out.
Request timed out.
Request timed out.
Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
Reply from 192.168.3.25: bytes=32 time=1ms TTL=255
Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=5ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=5ms TTL=255
Reply from 192.168.3.25: bytes=32 time=15ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=13ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
Request timed out.
Request timed out.
Reply from 192.168.3.10: Destination host unreachable.
Request timed out.
Reply from 192.168.3.25: bytes=32 time=5ms TTL=255
Reply from 192.168.3.25: bytes=32 time=5ms TTL=255

rstrouse, 2024-02-25:
This seems to indicate that either the power adapter is browning out or the connections are not quite correct.  Do you have another adapter?  Perhaps some pics of your connections are in order.  The pic in the Wiki is a different model ESP32 from the model you have.

luismalddonado, 2024-02-26:
Yes. I have tried several power adaptar and also tried with the 5V pin of the ESP32. Always same result. 

I'm not handy, but I guess PINS are correct as they work under some circustances. I have reviewed them several times.
I will try during this week other ESP32 models.

Thanks for you support and congratulations for this amazing project. Finally a really open solution for Somfy products!

luismalddonado, 2024-02-26:
ok. With a different ESP32 board is working perfectly!!!! This issue can be closed and thamks for the support

closed 2024-02-26 by luismalddonado