rstrouse/ESPSomfy-RTS #287: Alert pictogram on rolling shutter

gregoo31, 2024-02-26:
Good morning,

I have just noticed that I have a roller shutter with the pictogram "a danger sign and a sun". everything seems to be working normally. I couldn't find any explanations on the github. 

Do you know what this could come from?

Thanks in advance.

[image]

rstrouse, 2024-02-26:
If you do not have a wind sensor on that motor then you sent it a sun and wind sensor command from the virtual remote.   Go to the virtual remote select that motor from the dropdown and send it a sensor command with none of the checkboxes checked.

gregoo31, 2024-02-26:
Thank you, the pictogram is gone