rstrouse/ESPSomfy-RTS #288: Stumbled over SLZB-06: might this be used for ESPSomfy-RTS?

fft1010, 2024-02-26:
Hi,
I just stumbled over https://smlight.tech/product/slzb-06/ and wonder how are chances, that I could use this for ESPSomfy-RTS hardware instead of the things, that I did.

IMHO it might be much more accepted by the management :-) here in my house.

It has an ESP32 but I cannot judge, if the RTS sending part can be fulfilled with the hardware.

thanks
Juergen

rstrouse, 2024-02-26:
Those are zigbee.  Your motors would need to be zigbee motors not RTS.

fft1010, 2024-02-26:
Hi @rstrouse,
yes, I have seen it, but my idea is to flash this with your software. It has an ESP32, it has a WLAN and it has a 2.4MHz sender. As far as I understood Zigbee is "only" the protocol. I "just" wanted to use the hardware.

Any chance?

thanks
Juergen

rstrouse, 2024-02-26:
Here are some pics of my latest creation with a tiny ESP32-S3 and a 58x35x15mm enclosure from Amazon.

[image]
[image]

rstrouse, 2024-02-26:
@fft1010 that device will not work as the radio is not capable of producing the sub-ghz signals required.

rstrouse, 2024-03-09:
I am going to close this for now.

closed 2024-03-09 by rstrouse