rstrouse/ESPSomfy-RTS #290: MY simulation

rgr101, 2024-02-26:
### Describe you new feature you'd like

First of all, thank you so much for this great application!

We have a total of 22 RTS drives, two of which are shutters that cannot set a MY position, and this is very annoying. 
I was thinking that it would be relatively easy to implement a "pseudo-MY" function in ESPSomfyRTS for such drives by assigning a specific target position to the shutter as the MY position and simulating the MY behavior based on this value. 
This could even be used to retrofit awnings or blinds with a MY position. 
(Of course, something similar could also be realized individually in a higher level system like HA (or in my case Loxone) using the Target command, but this would require repeated additional efforts...)

By using Somfy, you're just used to their genious "MY comfort"... ;)

rstrouse, 2024-05-04:
I have added this function in the v2.4.3 pre-release.  There is now a checkbox on the shade definition for a simulated favorite.

rstrouse, 2024-05-11:
@rgr101 I have added this functionality to v2.4.3 and will be releasing it this weeken.

closed 2024-05-11 by rstrouse

rgr101, 2024-05-14:
Just installed - amazing !!!! This works really great !!! 👍🙏🙏🙏

Nickduino, 2024-05-17:
> EDIT: Because you didn’t ask it will allow you to set the my position using the remote as well by holding down the my button.

This is bloody brilliant!

rgr101, 2024-05-17:
> > EDIT: Because you didn’t ask it will allow you to set the my position using the remote as well by holding down the my button.
> 
> This is bloody brilliant!

Indeed!!

Nickduino, 2024-05-22:
> EDIT: Because you didn’t ask it will allow you to set the my position using the remote as well by holding down the my button.

...could the position be remote-dependant?

rstrouse, 2024-05-22:
> ...could the position be remote-dependant?

Because you asked. Sure!