rstrouse/ESPSomfy-RTS #291: Shutter group

carter013, 2024-02-27:
### Hardware

ESP32

### Firmware version

v2.3.1

### Application version

v2.4.0

### What happened? What did you expect to happen?

Hello,
I have 10 Somfy RTS roller shutters. I created groups of shutters to centralize a floor.
Since the latest updates (greater than v2.3.1), certain panes no longer animate in the pane group, but they interact independently. 

### How to reproduce it (step by step)

```bash
Each time I test an update, I have to revert to v2.3.1
```

rstrouse, 2024-02-27:
Zip up your backup and upload it here.  That is very odd.

carter013, 2024-02-27:
[ESPSomfyRTS 2024-02-26T18_14_45_v2.3.1.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/14423474/ESPSomfyRTS.2024-02-26T18_14_45_v2.3.1.zip)
Here is the file in functional version 2.3.1.
If I apply the update, the "Cuisne" pane does not play in the "1er étage" group

Thank you for your help

rstrouse, 2024-02-29:
I installed your backup on a test device and all 6 shades moved as expected on 2.4.0.

carter013, 2024-03-01:
Hello, thank you for your feedback. The shutters move well.
On the other hand, via the integration of Home Assistant the “Cuisine” component remains frozen. It does not move in the "1er étage" group.

rstrouse, 2024-03-03:
Are you setting the position from HA or are you sending up, down, my commands?  There is some weirdness in the cover group component for HA.

carter013, 2024-03-03:
I manage the shutters on HA and I only use the up/down. no position. Did you manage to reproduce the problem ?

rstrouse, 2024-03-03:
I did not are you using a service or are you using a card?

carter013, 2024-03-03:
I use the integration manually

rstrouse, 2024-03-03:
Do you mean you send a group command via a service.

carter013, 2024-03-03:
Not necessarily. Via HA integration, I retrieve the panes and groups created on the ESP. By simply using the pane in the integration, it does not interact with the group (HA and ESP).

rstrouse, 2024-03-03:
Alright so I ran it from a different dashboard control and the issue showed up.  If you install the pre-release v2.4.1 firmware the issue should go away.

carter013, 2024-03-03:
It's perfect, all panes/groups react correctly now !
Thank you very much for your help and involvement. 
Well done.

rstrouse, 2024-03-03:
Awesome!  All I needed was to get the right conditions to replicate it.  It looks like HA sends these one at a time for a cover group on some of the cards despite the fact that I have overridden the service.  That led me to realize that not all of the ids existed on the state object.  So the ESPSomfy RTS firmware was truncating the list of group members.

closed 2024-03-03 by rstrouse