rstrouse/ESPSomfy-RTS #295: Boot Inestability

luismalddonado, 2024-02-29:
### Hardware

ESP32

### Firmware version

2.4.0

### Application version

2.4.0

### What happened? What did you expect to happen?

When booting the ESP32 device "sometimes" it does not respond:
   1) Web interface does not fully load (however it loads partially but is not operable)
   2) When pinging the device:
           C:\Users\luism>ping -t 192.168.3.25
           Pinging 192.168.3.25 with 32 bytes of data:
           Request timed out.
           Request timed out.
           Request timed out.
           Reply from 192.168.3.25: bytes=32 time=12ms TTL=255
           Request timed out.
           Request timed out.
           Reply from 192.168.3.25: bytes=32 time=194ms TTL=255
           Request timed out.
           Request timed out.
           Reply from 192.168.3.25: bytes=32 time=1233ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
           Request timed out.
           Request timed out.
           Reply from 192.168.3.25: bytes=32 time=1282ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
           Request timed out.
           Request timed out.
           Reply from 192.168.3.25: bytes=32 time=1053ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=7ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=25ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=4ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=2ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=3ms TTL=255
           Reply from 192.168.3.25: bytes=32 time=2ms TTL=255   <<--Try to access web interface
           Request timed out.
           Request timed out.

NOTES:
        1) Unfortunatly when trying to debug, connecting to a USB Drive in a PC it always works well
         2) If I start the ESP32 with the CC1101 Transceiver disconnected (3v3 pin unplugged) and after start I connect it, it works well after disabling/enabling Radio in the Web Interfacee

I use a brand new Micro-USB cable and a brand new 2.4 A USB charger.

### How to reproduce it (step by step)

```bash
1. Reboot the device
2. Connected to the Web interface
```

### Logs

```bash
N/A. When connecting to a USB drive in a PC it allways works!
```

rstrouse, 2024-02-29:
The ESP32 libraries were bumped to 2.0.14 between those versions.  I also use AIMesh and do not see the issue on any of 10 devices but I will take a look at what is different in those versions.

luismalddonado, 2024-03-01:
> The ESP32 libraries were bumped to 2.0.14 between those versions. I also use AIMesh and do not see the issue on any of 10 devices but I will take a look at what is different in those versions.

Thanks. Looking forward to test new versions when available. Again thanks for this amazing project.

rstrouse, 2024-03-02:
I would be suspect of that power supply.  In some cases the power supply cannot keep up with the spikes that are created when the WiFi radio on the ESP32 is engaged.  This causes the ESP32 to panic reset because of a voltage drop on the EN pin.  Which looks a lot like the ping pattern above.  There were lots of changes to the core ESP32 libraries for the IDF which is why 1.7.2 which is based on v2.0.10 of the ESP32 core works in your testing.

There are lots of posts on many forums about adding a capacitor to the EN pin to smooth this out but there is another thing that I have run into with another user who had weird labeling on their board.  If you have a board like this do not use the GPIO11 pin as a ground.  It may work but it will be unreliable.

https://forum.arduino.cc/t/weird-gnd-in-esp32-gnd-or-gpio-11/958140

luismalddonado, 2024-03-02:
Yes I am using GPIO11 as ground for the antenna . I will leave my producction environment untouched with v1.7.2 (stable for the last 3 days). I ordered a new CC1101. Once it arrives I will do furthers tests.

rstrouse, 2024-03-02:
> Yes I am using GPIO11 as ground for the antenna .

That while electrically can be used for a ground it should not be used.  That pin is connected to SPI and could be browning out when the radio is initialized.   You should use an actual ground pin.

rstrouse, 2024-03-09:
@luismalddonado did you get your issues sorted?  From what I have read, that labeling of GPIO11 as GND is a screen printing error.  While when pulled low it does provide a path to ground but it also cannot handle load.

luismalddonado, 2024-03-10:
I just got today a second antena, tested with it without using GPIO11  fro GND and it works.  It would be great to change the "Suggested Pins" section for "ESP-WROOM-32/ESP32S Models" indicating this and the close this issue.

Thanks for the support

closed 2024-03-10 by luismalddonado