rstrouse/ESPSomfy-RTS #296: Network monitoring

rgr101, 2024-03-01:
### Describe you new feature you'd like

After I thought that my ESP was now running stable after having optimized the power supply, this morning I noticed that some shutters did not raise (by automation via Loxone/http). The ESP obviously didn't execute the commands as the concerning shades' GUI states were still "closed". Of course, this can have many causes, and so I would very much like to track the ESP's detailed activities in such situations. The detailed logging on the serial monitor is great for that, but the ESP is just no longer connected to the PC...

So, I would very much appreciate an option for redirecting the logging to a network monitor. 
A very simple and lean UDP implementation would be suitable for listening by a network monitor such as PacketSender and hopefully doesn't cost too much resources (and even doesn't bother the ESP's flash memory ;) )

I have already done this in my own projects and it works fine (but I don't dare to make a direct contribution to your sophisticated code...).

rstrouse, 2024-03-02:
What are you expecting to see here?  Short of the boot up sequence (which you wouldn't get anyway since the network would not yet be up) all of the status data is currently sent in json format over a socket connection.  This includes the current wifi data.

A likely cause for the http data not executing is because of the http client on the loxone.  Http is largely an async operation and depending on the HTTP client the socket is dedicated so sending another request before processing the last one is potentially problematic.  I had this issue with HA as it simply blew over the HTTP requests when I did not perform a blocking transaction.  

The other issue is that it takes some idle time for the radio to allow the motors to cool between frames and you can easily overwhelm the receiver on the motor.  The best way to reduce this traffic is to send group commands but don't forget motors do not understand the concept of seeking to a location. 

Are you using websockets on your Loxone?  If you do the status will always be up to date.

rgr101, 2024-03-02:
Ah I see. 

So far, I'm controlling the ESP by simple http://... shadeCommand requests via the Loxone virtual output port. In this regard, I'm driving "blind" with no feedback, so my goal was to collect feedback via the network monitor.
Unfortunately Loxone only supports tcp, udp, http and wol, but no websockets (websockets are supported by Loxone though, but only if Loxone is the server).

Yes, reducing the number of requests using groups might be a good idea - as I don't know how to do a blocking transaction in Loxone yet either.

Everything worked again this morning - that's really strange, and I don't like such sporadic problems at all as long as I can't understand them logically ;) 

btw - does the ESP fall into some kind of sleep mode (wifi?) after a certain idle time and then takes some time to wake up? (I think I had the impression that sometimes - after a long period of inactivity - the first command -even from its own webgui -  was executed with a delay of several seconds but all subsequent ones immediately)

If so, I could relatively easily implement a wake-up in Loxone and send the actual requests with a time delay.

rstrouse, 2024-03-02:
I have disabled sleep mode for WiFi.  I don't have any experience with loxone.  But are you saying they implement websockets but you cannot listen on an open socket?  I do see references on the interweb though of what is referred to as a loxone-websocket-client.

If you have battery motors then it is very likely that they go to sleep.   For those give it a couple of repeats in the shade definition.

rgr101, 2024-03-03:
Wifi sleep mode disabled:👍

I’m afraid my knowledge in websockets is not sufficient - as far as I understand it‘s quite complicated to establish a websocket connection to the Loxone Miniserver and it seems to having to be initiated always by the counterpart 

No, I only have 230V motors. 

btw - tonight everything went well again, closing 20 shades by 20 single requests 😇, all fired by Loxone within the same second.

rstrouse, 2024-03-03:
Like I said, I have no real knowledge of Loxone.  Websockets are the most straightforward way of passing data so I would be shocked that there is no off the shelf socket client solution for events.   Your other options include MQTT or even polling but polling is terribly inefficient.

rgr101, 2024-03-03:
Yes, I find that strange, too. And you‘ll again be shocked by hearing that Loxone doesn‘t natively support MQTT. They are pursuing an increasingly proprietary business policy and opposing many  open standards (or offer wickedly expensive gateways for some) which makes me increasingly uncomfortable...

HA is relatively new to me and I’m pretty impressed, so I'm actually thinking about switching from Loxone to HA in the future; on the other hand, I'm a bit reluctant to move away from an unpleasantly proprietary but super stable system and embark on an "unknown adventure"... But maybe I'll find a "gentle" smooth migration path where both systems coexist for a while. I have seen that there is an HA plugin for Loxone...
My problem is that I'm aiming more for a reduction in complexity, as my grown environment is already too heterogeneous and maintenance-unfriendly (Loxone, KNX, Homebridge, Mediola, ESPSomfyRTS, my own ESPs, Somfy, Elro, Intertechno...)! But I already have some ideas for streamlining, and my hope is that the incredible  flexibility of HA could support me in this endeavor ?!?

rstrouse, 2024-03-04:
I have been at the automation game for a while.  HA really does pull together disparate options into a single platform. Almost everything in my home is automated inside and out.

rgr101, 2024-03-04:
Sounds great!  
Would you think I would be well served with a "Home Assistant Green" device to start with? 
An out-of-the-box solution seems attractive to me, especially for the central unit 😇 - However, I still don't have a feeling for the actual resource requirements.

rstrouse, 2024-03-04:
I went with a fanless i7 pc home server running proxmox. That being said I have a lot going on.  Irrigation, pool, wine cellar, garages, alam, climate, media, and a whole lot of switches and plugs.  Lots of folks run HA on pretty modest hardware.

rstrouse, 2024-03-04:
Btw I the vm on the home server is way more power than is needed. The resources available to the green are about what I have assigned to the VM.

rgr101, 2024-03-04:
Great! This is exactly what I wanted to hear 😁

rstrouse, 2024-03-09:
I am going to close this.  I assume you are on your HA journey.  Good luck.

closed 2024-03-09 by rstrouse

rgr101, 2024-03-10:
Exactly: I grabbed a "Green" box - works really fine. 
But the HA journey is arduous, esp. the first steps, as Loxone has conditioned me to graphical block programming, so I'm far away from textual coding and never used YAML... (it's hard to find a YAML-HA language reference!?!)
All the more by myself not being the greatest coder (more an architect)... Luckily ChatGPT (don't laugh!) is a good facilitator and teacher, even if its codes often is the biggest nonsense - but also this is a good training...

I already succeeded in integrating your ESPSomfy HA integration - working out-of-the-box like magic !!! 
Thanks again for your great contribution!
Now my next challenge is to implement the automation stuff...