rstrouse/ESPSomfy-RTS #298: Version 2.4.x

harald1952, 2024-03-03:
### Hardware

ESP32

### Firmware version

V2.4.0

### Application version

V2.4.0

### What happened? What did you expect to happen?

Bei der Installation von 2.4.0 kein stabieler Boot vorgang. Keine möglichkeit den Hotspot zu erreichen. Startvorgang wiederholt sich in einer endlosschleife nur v2.3.0 läuft stabil

### How to reproduce it (step by step)

```bash
nach instalation kein start bis zum hotspot
```

rstrouse, 2024-03-03:
Please provide more information about what model of ESP32 device you are provisioning.  The 2.4.0 firmware has been installed many times since its release.  Typically if the boot up is going into a constant loop it is because the wrong firmware was installed for the ESP32 device.

If you can please provide your serial logs.

PauloGithub33, 2024-03-05:
I think this is where I should have asked my question, sorry for the duplicate 

Hello, First of all, congratulations for the project, the follow-up and the didactic side. It's very impressive. 

I'm starting with home assistant, a rasberry5 and RTS shutters. I encounter a problem during the integration with HACS <Integration rstrouse/ESPSomfy-RTS> Repository structure for v2.4.0 is not compliant A new version problem perhaps or I missed a step? ESP32vroom and SomfyController.onboard.esp32   

thank you for all your work, I admire it. I only know how to copy

rstrouse, 2024-03-05:
This is the ESP32 firmware repository.   There is another repository for the HA integration.

https://github.com/rstrouse/ESPSomfy-RTS-HA

rstrouse, 2024-03-09:
I think we have all the issues resolve in this submittal.  If you are still having issues feel free to reopen this issue or create a new one.

closed 2024-03-09 by rstrouse