rstrouse/ESPSomfy-RTS #299: Somfy IO

harald1952, 2024-03-03:
### Describe you new feature you'd like

ist es beabsichtigt da immer mehr Somfy io  mit 868Mhz ausgeliefert wird espsomfy hierfür zu ertüchtigen bzw. wo liegt hier ein Problem vor. der CC1101 ist ja auch für dieses Frequenzband zu bekommen.

rstrouse, 2024-03-03:
Please submit your issues in English.  However in this case I am pretty sure I know what you are asking for.   I do not have any IO homecontrol devices and will probably never own any.  The notion of cloud provisioning for appliances in my home leaves a really bad taste in my mouth.  As consumers we need to question why a doorbell, vacuum, thermostat, garage door, refrigerator, or roller shade needs to have dependency to a cloud server.

There was an effort a while ago to take this on but I really don't have any need for it.
https://github.com/Velocet/iown-homecontrol

rstrouse, 2024-03-09:
I'm sorry but I am not equipped to develop such a solution as I do not have any of that hardware.  If I did I would probably be fielding questions about RTS support :)