rstrouse/ESPSomfy-RTS #301: Homekit

TibsD, 2024-03-08:
### Describe you new feature you'd like

Maybe a long shot but at the moment I am also using the Home Assistant integration to get my RTS blinds into Homekit.
Is there a possibiliy to get a native Homekit support? A bit like the Mongoose OS for the Shelly devices?

rstrouse, 2024-03-08:
Yeah I wouldn't get your hopes up on that one.   I also use the home assistant integration and it picks where our tiny little MCU ends.  As it stands there is almost no code space left on the device, so handling all of the last mile in Home Assistant is kind of where we are at right now.

rstrouse, 2024-03-09:
Yeah I went back and looked at several solutions that create a Homekit device on ESP32.  All of them are hacks at this point since Apple made sure you adopt their sdk and subsequently invest in it.   Interestingly, the hub devices don't seem to have similar restrictions.

Either way it would be a lot of work and result in a lesser solution.   I am going to close this for now.

closed 2024-03-09 by rstrouse