rstrouse/ESPSomfy-RTS #304: Index.html not found

technicsdj, 2024-03-09:
### Hardware

ESP32

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

hello i have now tried to flash the latest firmware on my esp32 microcontroller. with the tasmota esp flasher i always get the error index.html not found. i have tried several flashers, but none of them work properly. with the esp flashtool, as described in the instructions, the esp controller does not even boot up. the bootled always just flashes. i have tried all the firmware versions offered with both flashers. what am i doing wrong? greetings and thanks

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

### Logs

```bash
Using 'COM5' as serial port.
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Connecting.........

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-D0WDQ6 (revision v1.0)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1500000
Changed.
 - Flash Size: 4MB
Creating esp32 image...
Successfully created esp32 image.
Erasing flash (this may take a while)...
Chip erase completed successfully in 14.1s
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000d0fff...
Flash will be erased from 0x000e0000 to 0x00216fff...
Compressed 17920 bytes to 12440...
Writing at 0x00001000... (100 %)
Wrote 17920 bytes (12440 compressed) at 0x00001000 in 0.3 seconds (effective 464.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 134...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (134 compressed) at 0x00008000 in 0.0 seconds (effective 545.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 759.7 kbit/s)...
Hash of data verified.
Compressed 786672 bytes to 560794...
Writing at 0x00010000... (2 %)
Writing at 0x00016962... (5 %)
Writing at 0x0001f2d3... (8 %)
Writing at 0x00029baf... (11 %)
Writing at 0x0002f503... (14 %)
Writing at 0x00034a9a... (17 %)
Writing at 0x00039a30... (20 %)
Writing at 0x0003ebf9... (22 %)
Writing at 0x00043c0d... (25 %)
Writing at 0x00048b3b... (28 %)
Writing at 0x0004d8d7... (31 %)
Writing at 0x00052554... (34 %)
Writing at 0x0005732a... (37 %)
Writing at 0x0005c354... (40 %)
Writing at 0x00061aaf... (42 %)
Writing at 0x00067ee9... (45 %)
Writing at 0x0006d786... (48 %)
Writing at 0x00072b89... (51 %)
Writing at 0x00077fa6... (54 %)
Writing at 0x0007d255... (57 %)
Writing at 0x00082677... (60 %)
Writing at 0x00087d9a... (62 %)
Writing at 0x0008ce46... (65 %)
Writing at 0x0009254c... (68 %)
Writing at 0x000977c8... (71 %)
Writing at 0x0009cc32... (74 %)
Writing at 0x000a2054... (77 %)
Writing at 0x000a7632... (80 %)
Writing at 0x000aceef... (82 %)
Writing at 0x000b2b51... (85 %)
Writing at 0x000b86f8... (88 %)
Writing at 0x000bdab0... (91 %)
Writing at 0x000c3220... (94 %)
Writing at 0x000c8d6f... (97 %)
Writing at 0x000cea8b... (100 %)
Wrote 786672 bytes (560794 compressed) at 0x00010000 in 8.1 seconds (effective 779.7 kbit/s)...
Hash of data verified.
Compressed 1271024 bytes to 796998...
Writing at 0x000e0000... (2 %)
Writing at 0x000eb8f6... (4 %)
Writing at 0x000f881a... (6 %)
Writing at 0x00102c7f... (8 %)
Writing at 0x0010ce78... (10 %)
Writing at 0x0011dc12... (12 %)
Writing at 0x001236e5... (14 %)
Writing at 0x00129010... (16 %)
Writing at 0x0012ebdd... (18 %)
Writing at 0x00134941... (20 %)
Writing at 0x0013af41... (22 %)
Writing at 0x0014263d... (24 %)
Writing at 0x00148700... (26 %)
Writing at 0x0014e457... (28 %)
Writing at 0x00154417... (30 %)
Writing at 0x0015a126... (32 %)
Writing at 0x0015f8df... (34 %)
Writing at 0x001658ba... (36 %)
Writing at 0x0016c1a6... (38 %)
Writing at 0x00171575... (40 %)
Writing at 0x001767d8... (42 %)
Writing at 0x0017bb2f... (44 %)
Writing at 0x00180f4c... (46 %)
Writing at 0x001861d1... (48 %)
Writing at 0x0018b32e... (51 %)
Writing at 0x00190897... (53 %)
Writing at 0x00195795... (55 %)
Writing at 0x0019aebe... (57 %)
Writing at 0x0019ff74... (59 %)
Writing at 0x001a51d6... (61 %)
Writing at 0x001aa6be... (63 %)
Writing at 0x001b049a... (65 %)
Writing at 0x001b5b01... (67 %)
Writing at 0x001bac24... (69 %)
Writing at 0x001c011b... (71 %)
Writing at 0x001c53c1... (73 %)
Writing at 0x001ca823... (75 %)
Writing at 0x001cfd23... (77 %)
Writing at 0x001d543c... (79 %)
Writing at 0x001dab5b... (81 %)
Writing at 0x001e09e7... (83 %)
Writing at 0x001e64a6... (85 %)
Writing at 0x001ed374... (87 %)
Writing at 0x001f4f20... (89 %)
Writing at 0x001fd0ea... (91 %)
Writing at 0x00202199... (93 %)
Writing at 0x00207ec2... (95 %)
Writing at 0x0020d316... (97 %)
Writing at 0x002129a5... (100 %)
Wrote 1271024 bytes (796998 compressed) at 0x000e0000 in 13.0 seconds (effective 780.6 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Hard resetting via RTS pin...
Done! Flashing is complete!

Showing logs:
[22:01:10]le System...
[22:01:10]E (9) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1347:error: Corrupted dir pair at {0x0, 0x1}
[22:01:10]
[22:01:10]E (11) esp_littlefs: mount failed,  (-84)
[22:01:10]E (14) esp_littlefs: Failed to initialize LittleFS
[22:01:10]Error mounting file system
[22:01:10]Chip Model ESP32-
[22:01:10]Preference IP Free Entries: 621
[22:01:15]SECURITY   Type:0 Username:[] Password:[] Pin:[] Permissions:0
[22:01:15]Connection Type: 1
[22:01:15]NTP Settings 
[22:01:15]pool.ntp.org TZ:
[22:01:15]WIFI Settings
[22:01:15] SSID: [] PassPhrase: []
[22:01:15]
[22:01:15]Launching web server...
[22:01:15]Creating Web MicroServices...
[22:01:16]WiFi Mode: 0
[22:01:19]Scanned 5 Networks...
[22:01:19] 0: XXXXX
[22:01:19] 1: XXXXX
[22:01:19] 2: XXXXX
[22:01:19] 3: XXXXX
[22:01:19] 4: XXXXX
[22:01:19]Socket Server Started...
[22:01:19]Max Heap: 110580
[22:01:19]Free Heap: 186828
[22:01:19]Min Heap: 184728
[22:01:19]
[22:01:19]Turning the HotSpot On
[22:01:19]WiFi AP Started
[22:01:19]Initializing AP for credentials modification
[22:01:19]
[22:01:19]SoftAP IP: WiFi AP Stopped
[22:01:19]192.168.4.WiFi AP Started
[22:01:19]1
```

rstrouse, 2024-03-09:
If this is a new onboard download the onboard file again and flash it again.  It looks like the littlefs partition is corrupted.  If you are using one of the espressif tools this can happen when the serial port write is corrupted.  I would suggest you download the onboard file again and use the web.esphome.io website.   Also please make sure you download the proper onboard file.

technicsdj, 2024-03-09:
Hey,
thanks for the idea. i have tested it but i have the same problem as with all flashers, that the bootled of the controller is only lit. the log shows the following errors:

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371

rstrouse, 2024-03-09:
That tells me that either the flash chip is broken or you are not using the correct SomfyController.**onboard**.esp32**xx**.bin.zip file.  The watchdog timer resetting like that is typically because the main bootloader cannot find the proper code to execute.

technicsdj, 2024-03-09:
i am using the following microcontroller. 
https://www.az-delivery.de/en/products/esp32-developmentboard
i can rule out that it is broken. i have tried a new one and i get the same errors. can you tell me which is the correct firmware?

rstrouse, 2024-03-09:
Ok so download the zip file below and extract it to a directory on your computer.

https://github.com/rstrouse/ESPSomfy-RTS/releases/download/v2.4.1/SomfyController.onboard.esp32.bin.zip

1. Using the web.esphome.io website click on the connect button.
2. Click on the `Install` button.  Do not click the PREPARE FOR FIRST USE button.
3. Select the extracted file `SomfyController.onboard.esp32.bin`  and press INSTALL

When it finishes installing the binaries you should remove power from the board and reboot it.   After it has completely booted you should see an ESPSomfy RTS access point in your wifi list.

technicsdj, 2024-03-09:
thanks for the instructions, i get the same error that the index.html cannot be found.

rstrouse, 2024-03-09:
I decided to flash another device with onboard firmware and it worked just fine.  What is the history with this board?  Is it currently hooked up to the radio?  If it is please remove the radio from all the pins.  That way we can make sure there isn't a wiring issue with the radio.

technicsdj, 2024-03-10:
there is no radio connected to the microcontroller. when the radio is connected, it cannot be flashed at all. i have now taken another brand new controller and i get the same errors. i am beginning to think it is the microcontroller that is not compatible with the software. i simply flashed a standard tasmota software on the controller and it works perfectly.

Errorlog new microcontroller:
Turning the HotSpot On
WiFi AP Started
Initializing AP for credentials modification

SoftAP IP: WiFi AP Stopped
192.168.4WiFi AP Started
.1
.........1Loading file /index.html
111111....................................................................................

rstrouse, 2024-03-10:
That is very odd.  When you reboot it while connected to the serial log do you still see that it cannot mount littlefs.  

This is the sequence from your log above that indicates that it cannot initialize the filesystem.
```
[22:01:10]E (9) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1347:error: Corrupted dir pair at {0x0, 0x1}
[22:01:10]
[22:01:10]E (11) esp_littlefs: mount failed,  (-84)
[22:01:10]E (14) esp_littlefs: Failed to initialize LittleFS
[22:01:10]Error mounting file system
```

If you are not getting that then hard refresh your browser to make sure we are not just getting cache corruption from the browser.  If this is chrome then hold down the control key and press f5 if it is Safari then you will need to clear the browser cache for 192.168.4.1.

Likely the reason why it is working fine with Tasmota is because it does not use the littlefs filesystem and therefore doesn't create a partition for it.  In looking at those docs it creates a completely different filesystem.

rstrouse, 2024-03-10:
Awesome!

rstrouse, 2024-03-11:
I am going to close this.  If you have other problems open a new issue.

closed 2024-03-11 by rstrouse