rstrouse/ESPSomfy-RTS #307: Centralis RTS manual remote controls no longer work after pairing

59marcopolo, 2024-03-20:
### Hardware

ESP32-C3

### Firmware version

v2.4.0

### Application version

V2.4.0

### What happened? What did you expect to happen?

When I press a button on the Centralis RTS remote, I can see in the web interface that the action is seen by the software, but the motor doesn't move.

### How to reproduce it (step by step)

```bash
Hello,
I have installed ESPSomfy RTS, firmware V2.4.0. It works perfectly with my 3 shutters and my terrace awning.
However, I've lost the use of my 3 Somfy Centralis RTS remote controls, following the pairing.
When I press a button on the Centralis RTS remote, I can see in the web interface that the action is seen by the software, but the motor doesn't move.

Only the Somfy TELIS 1 RTS PURE remote control for the awning still works manually.

Thank you for your help.
```

rstrouse, 2024-03-20:
There are really only two reasons that this can occur.  The most common cause is because the same address was used for one of your ESPSomfy RTS shade definitions as the remote and the other is that the remote was inadvertently paired while pairing motors.  If the memory was open on a motor and you pressed the prog button on that remote it will remove the reference from the motor.  It is very likely either one of the above.

59marcopolo, 2024-03-20:
Can I pair my remote again via the ESPSomfy software, or do I have to go through the "unplug motor, wait 2s, plug in motor, wait 7s ..." procedure?

rstrouse, 2024-03-20:
Yes go to the virtual remote tab, select your motor from the dropdown, then long press on the prog button until the motor jogs.  Then short press the prog button on your remote.   It should jog again.

Check to see if the motor then responds to your remote.  If it does then you are good to go.  If it does not do the procedure above 1 more time and test the remote.   If you have to do it twice the first time successfully removed the remote from the memory.  The second time added it back in.

If this does not work the motor has blacklisted the remote and you will need to go back through the factory reset procedure.

59marcopolo, 2024-03-21:
Hello,
I followed your procedure. The remote controls are operational again.
But when I switch from Web portal mode to remote control mode, I have to press the buttons a lot (between 10 and 20 times) for the first action to take effect. Then it's normal.
The same applies when I switch from remote control mode to web portal.

59marcopolo, 2024-03-21:
I've also seen indications that I didn't have before on the web portal, a danger sign on two of the three shutters, and a sun on one of the shutters.
[image]

rstrouse, 2024-03-22:
Please go into the Virtual remote, select the decorated shades from the dropdown and send a sensor command without any of the checkboxes checked.   The warning icon is because the motor thinks there is a wind condition that should retract the motor and the sun indicates a sunny condition.

In messing around with the Virtual Remote you sent signals as if there is a connected Eolis.

59marcopolo, 2024-03-23:
> Please go into the Virtual remote, select the decorated shades from the dropdown and send a sensor command without any of the checkboxes checked. The warning icon is because the motor thinks there [...]

Okay, got it, I was able to delete the danger and sun signs, thanks.

59marcopolo, 2024-03-23:
However, I still have the problem of switching from Web mode to manual mode.
I have to press a large number of times (between 15 and 20) on one of the control buttons to regain control in manual mode.
And vice versa.

I have this problem with the 3 Centralis RTS roller shutter remote controls.

I don't have the problem with the TELIS 1 RTS PURE remote control for the terrace awning.

rstrouse, 2024-03-23:
By Manual mode I assume you mean from the remote.  This really looks like the address of your remote is being used by one of your shade definitions.  The number of times you have to press on the Centralis is likely the number of times you moved them with ESPSomfy RTS.  

First open up the Radio -> Log tab in ESPSomfy RTS and press each of your remotes to record the addresses.

[image]

Then open up the Somfy -> Shades tab and make sure none of the recorded addresses from the log tab match any of the addresses assigned to the ESPSomfy RTS shade definitions.  In the example I am showing you the 1147001 will have issues.
 
[image]

59marcopolo, 2024-03-23:
You're right, I have the same addresses in the log as in the Shades tab !
This is for the 3 remotes.

59marcopolo, 2024-03-23:
I reread your documentation carefully, and it was I who made the mistake. I changed the Remote Address when I should have left it as calculated.
I've deleted and recreated the 3 shutters, and everything works perfectly now.
I apologize for the inconvenience.
You can close my request.
Thanks

rstrouse, 2024-03-23:
Awesome!   Glad we got you going.

closed 2024-03-23 by rstrouse