rstrouse/ESPSomfy-RTS #308: Documentation - no existing remote

Dinth, 2024-03-20:
### Describe you new feature you'd like

Is it possible to pair ESPSomfy-RTS with blinds without an existing remote or migrate the configuration from RFLink? Until now i have been controlling my curtains using RFLink device and the original remote broke years ago. I want to migrate to ESPSomfy-RTS, but havent found any information how to actually do that.

rstrouse, 2024-03-20:
Yes it is fairly easy to do.  I have never used RFLink so if there is a way to open the memory with a prog command from the RFLink then you simply treat it as a remote.  

If there is not then you will need to factory reset the blinds and perform the initial pairing sequence using the virtual remote.  First, create a shade definition and then click on the Virtual Remote tab.  Select the new shade definition and go through the initial pairing steps for your motor.  Every remote button and button combination you require exists on this tab.  

If the RFLink is capable of opening the memory with a long press prog command then you simply pair it as if it was the original remote.

After you pair the motor to ESPSomfy RTS do not forget to link the RFLink to the ESPSomfy RTS shade definition in the linked remotes list.  Then you can use either method to control your blinds.

Dinth, 2024-03-20:
RFLink is most likely capable of sending the prog command, but i dont have the required command to send and i may be wrong, but as far as i remember pairing process was tricky even with the original remote as i have multiple motors plugged in directly to mains. 
Would it not be possible to just take the RFLink device ID (which is 0A0A0A in my case) and clone it?

rstrouse, 2024-03-20:
If you do that then your RFLink will stop working and if you use it it will cause ESPSomfy RTS to also stop working.  However, we may be able to get you through this and still have your cake and eat it too.

### Step 1:
Open the logs screen and observe/record the address and rolling code sent by RFLink

### Step 2:
Create a new shade definition in ESPSomfy RTS and call it RFLink.  Use the address that you received from RFLink.

### Step 3:
After adding the shade definition click the set rolling code button and set the rolling code to the address from RFLink

### Step 4:
Create another shade definition for your blind.  DO NOT USE THE ADDRESS FROM RFLink.  Let ESPSomfy RTS assign this address.

### Step 5:
Open the virtual remote tab and select RFLink from the dropdown.  Then press and hold the prog button until the blind jogs.

### Step 6:
Edit the blind you created and click pair and follow the instructions.  The blind should already be in pairing mode.

### Step 7:
Delete the RFLink shade definition.  You should be able to use the RFLink interface after sending a few commands from it and the rolling code catches up.

rstrouse, 2024-04-11:
@Dinth I assume you have gotten sorted with this so I am going to close it.

closed 2024-04-11 by rstrouse