rstrouse/ESPSomfy-RTS #309: No boot or logs on first install

AlexanderMohan, 2024-03-20:
### Hardware

ESP32-C3

### Firmware version

2.4.0

### Application version

2.4.0

### What happened? What did you expect to happen?

installed via esphome web installer. no logs or AP starts

### How to reproduce it (step by step)

```bash
1. install firmware
2. observe no activity
```

### Logs

```bash
ESP-ROM:esp32c3-api1-20210207
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4207ef36
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x25f4
entry 0x403cc710
```

rstrouse, 2024-03-20:
You must download the proper bin file.  The message above indicates that you installed the wrong SomfyController.onboard.espXX.bin file.  Download the ,zip file circled above, extract it, then install the `SomfyController.onboard.esp32c3.bin` file from within this archive.  Complete instructions are in the wiki https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware.

[image]

AlexanderMohan, 2024-03-20:
Hi, I definitely installed that one. I've just redone it and it's still not working.

rstrouse, 2024-03-20:
Which C3 do you have?  Does it require jumpers or pins to be enabled to be set in order to flash and did the flash say it was successful?  If it did remove power to let it reset.

AlexanderMohan, 2024-03-20:
I have the C3 super mini, I did let it reset after flashing - it has a boot button and other firmwares installed fine.

rstrouse, 2024-03-20:
Please remove power.  I just flashed a C3 with that firmware and it required removal of power to set the jtag properly then booted happily in AP mode.  If I just soft reset I got your results.

rstrouse, 2024-03-23:
Please install the v2.4.1 release.  I have added some settings to the C3 build to ensure the CDC gets initialized for the serial port on boot.  Some boards do not initialize this by default.

EnricoBettella, 2024-03-26:
I also have a C3 super mini board and couldn’t get the firmware working, even though I tried several times. I tried to unplug the power too. It simply doesn’t start the AP even if the flasher says completed succesfully.

AlexanderMohan, 2024-03-26:
I got it working on an alternative version of the c3 super mini. I wonder if there’s some under-documented difference between the two? Will share links soon.

rstrouse, 2024-03-26:
If you provide the links to the working one and the non-working one perhaps we can see the difference.  Mine that works great happens to be the Seeed Studio version.

EnricoBettella, 2024-03-26:
**Definitely my bad**: I was making a mistake during the flashing process. The “SomfyController.onboard.esp32c3.bin.zip” file couldn’t be unzipped by macOS so I thought it was the case just to rename it removing the .zip end part. ESPHome didn’t give any error during flashing so I didn’t worry about it.
I managed to unzip with a third party app and then everything went flawlessly.
So at least now I can confirm that the ESP32 C3 SuperMini that I bought here https://a.aliexpress.com/_EyGKKS1 works!

rstrouse, 2024-04-08:
I am closing this for now.  Glad you got it working.