rstrouse/ESPSomfy-RTS #310: Restart of the module

011V32, 2024-03-23:
### Describe you new feature you'd like

It would be great if you could add a restart button so that the system could be brought to the boot-up state. (And so that it gets a new IP, possible memory leaks can be fixed, etc.)
Ideally, this should not only be possible via the web interface, but also be triggered externally (e.g. via HomeAssistant), so that you can, for example, restart it cyclically..
Thanks.

rstrouse, 2024-03-23:
On the DHCP/Static IP tab if you temporarily uncheck the Acquire IP Address automatically checkbox the current IP address is placed in the Static IP field.

rstrouse, 2024-03-23:
I published 2.4.1 and the IP address is now prominently displayed on the DHCP tab.

011V32, 2024-03-24:
I couldn't see it. Even if I uncheck the DHCP option. 🙃
[image]
[image]

rstrouse, 2024-03-24:
I am not sure how that could be possible.  Please clear your browser cache.

011V32, 2024-03-26:
I tried. The InPrivate window didn't help either. Neither on the Mac, nor on the PC or cell phone & tablet... 😞

rstrouse, 2024-03-26:
Are you seeing any errors in the browser console?

Jumperm, 2024-03-28:
Today I want to check the browser console in Chrome and today is the first day I see the IP. In Safari (macOS/iOS), chrome ... curious!

rstrouse, 2024-03-28:
Most likely, the cache expired for the client files in the browser.

011V32, 2024-04-02:
Where I have to look?
[image]

rstrouse, 2024-04-04:
That I didn’t expect it looks like the Ethernet library isn’t returning a current ip address.

rstrouse, 2024-04-10:
If you install the v2.4.2 pre-release I believe I found the cause of this.   As it turns out the IP address sometimes does not get assigned until after the uplink reports connected.

closed 2024-04-11 by 011V32