rstrouse/ESPSomfy-RTS #311: Unable to update from Home Assistant

FlaMike, 2024-03-24:
### Hardware

ESP32

### Firmware version

2.4.0

### Application version

2.4.0

### What happened? What did you expect to happen?

I was not able to update from 2.4.0 to 2.4.1 from within Home Assistant.  The update showed up in HA and I clicked on the update button in the integration, but nothing happened.  Reproduced several times.  I went to the device & clicked github update, it backed up the files but did not update the firmware or the application (it showed "Main" for the version to retrieve from github).  Tried  a second time, it displayed 2.4.1 as the version, backed up the files & then proceeded to update both the firmware & the app successfully.

If I recall correctly, for the update from 2.3.3 to version 2.4.0 I had the same challenge within HA & had to go to the device to update it.  Back when updating the firmware & app were introduced as HA integration options I had no problem updating from within HA.

I am currently running HA Core 2024.2.5 and front end 20240207.1 in a docker container (no HAOS).

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-03-24:
If it shows only main then it is having trouble getting to GitHub at the moment.

FlaMike, 2024-03-25:
OK.  Strange because amazingly, I was not experiencing any internet problems at the time, but perhaps github's response was too slow.

Thanks for your quick reply!

rstrouse, 2024-03-25:
GitHub itself can have fuzzy edges where services become unavailable at times.  I assume you finally got your update done.

FlaMike, 2024-03-25:
I completed the firmware & app updates yesterday & just did the HA integration update.  All good here!