rstrouse/ESPSomfy-RTS #313: Lost all config after an update

Nickduino, 2024-03-31:
### Hardware

ESP32-C3

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

I lost all my rooms and shades after the update. I expected them to be restored.

I'm an idiot, I should have done a manual backup but I was on the Home Assistant app, not my PC, so...

### How to reproduce it (step by step)

```bash
1. Open Home Assistant settings
2. Click "see all updates"
3. Select ESPSomfy RTS firmware and update
4. Once updated, see that there are no blinds and no rooms in your ESPSomfy RTS install
```

Nickduino, 2024-03-31:
> I'm an idiot, I should have done a manual backup but I was on the Home Assistant app, not my PC, so...

I thought I had an older one but I can't find it.
**"Create backup" was ticked on the Home Assistant dialog box. Did HA do a backup of my shades? How can I access it?**

Nickduino, 2024-03-31:
I read the other similar issues but I don't have a backup file to try and restore the shades

>  **Did HA do a backup of my shades? How can I access it?**

I'm a HA noob, should the backup be here?
[image]

Or is there a different backup directory for HACS?

rstrouse, 2024-03-31:
Yes the backup is in the `config/ESPSomfyRTS_XXXXXX` where Xs = the chip id.

rstrouse, 2024-03-31:
You can use the file editor in HA to download this file.  If you click the folder icon you will see the directory

[image]

Nickduino, 2024-03-31:
> Yes the backup is in the `config/ESPSomfyRTS_XXXXXX` where Xs = the chip id.

> You can use the file editor in HA to download this file. If you click the folder icon you will see the directory

Thanks! I restored all shades, rooms and groups :-)

Maybe you could mention the HA backup folder somewhere (like on the firmware tab of the GUI)?

closed 2024-03-31 by Nickduino

rstrouse, 2024-03-31:
> Maybe you could mention the HA backup folder somewhere (like on the firmware tab of the GUI)?

Eventually I am going to add a UI to HA to allow you to list and restore a backup file.  In the meantime that is not such a bad idea.