rstrouse/ESPSomfy-RTS #314: Stop button often doesn't work, when Up and Down buttons work great

AAfetisov, 2024-03-31:
### Hardware

ESP32-S3

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

Stop button often doesn't work, when Up and Down buttons work great. I sometimes click Stop tons of times with 0 effect. The remote control's Stop always work.
why could that be?

### How to reproduce it (step by step)

```bash
press Up or Down, press Stop. Sometimes it wont work no matter how many times you press it and the motor will be running for another 10sec until it stops because of the timer(probably). Tried this on esp32-s3 and esp32-wroom-32
```

Nickduino, 2024-03-31:
But why would this impact the stop button more than the two others?

I too had the feeling the stop button was less responsive than the up and down but I thought it was just in my head.

rstrouse, 2024-03-31:
I have seen a few instances where this happens.  I have a theory that it is related to interference created by the running motor.

Nickduino, 2024-03-31:
> I have a theory that it is related to interference created by the running motor.

But these interferences would jam the up or down signal as well (when it's running, you can change its direction but not stop it).

rstrouse, 2024-03-31:
Perhaps.  It is just a theory.   On my motors I have not witnessed any difference between sending My and any other command.  It all routes through the same place for the command interface.  The motors themselves do listen for long press and repeats on the My command though while they typically do not for up/down.

Nickduino, 2024-03-31:
> The motors themselves do listen for long press and repeats on the My command though while they typically do not for up/down.

I don't think my 30 year old motors do listen for a **my** command. And again, I've not seen that behaviour with my physical remotes or script.

But that might be in my head. It's just that seeing someone else saw that behaviour made me think there was something to it.

AAfetisov, 2024-04-01:
> Increase the number of times that the commands are repeated to the motor. You will find this dropdown when you edit the shade definition.

I tried that. I'v set it to 5 repeats. Tried 10 and 20 too. Could it be that the app freezes sometimes after receiving Up or Down? Because I also had ui giving me very long or even endless spinning balls when clicking on my drapes.

Nickduino, 2024-04-01:
> Could it be that the app freezes sometimes after receiving Up or Down?

If you move the shutter with the physical remote, does the stop button works on ESPSomfy?

rstrouse, 2024-04-02:
Please test what @Nickduino said.   

Also try it with the IP address if you are currently using the DNS name for the device.  Depending on your network it may not be efficiently resolving mDNS.  In that instance the browser console will have a message indicating that it cannot resolve host name.  

Depending on your network mDNS issues can sometimes be resolved by reserving an IP address and supplying the hostname in your router.

AAfetisov, 2024-04-15:
> Also try it with the IP address if you are currently using the DNS name for the device. Depending on your network it may not be efficiently resolving mDNS. In that instance the browser console will have a message indicating that it cannot resolve host name.

o_O, this seems to be exactly the case. It works flawless if I use static ip instead of a domain name.

ps
Sorry for the delay

closed 2024-04-15 by AAfetisov