rstrouse/ESPSomfy-RTS #315: Can't link remote

cvhoang, 2024-04-01:
### Hardware

ESP32

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

I've managed to pair 4 shades to ESPSomfy-RTS and can control them without issues. However, I can't seem to link remotes to the shades. The "Link Remote" screen stays on and the shade doesn't close after pressing the Telis 4 buttons. Does it mean the RX pin is not correct? 

In the "Logs" tab only displays activities if I use "Scan Frequency" function. Otherwise it won't record any entries at all. 

Thank you.

### How to reproduce it (step by step)

```bash
1. Click on "Link Remote"
2. Press one of the buttons on Telis 4 remote
3. The "Link Remote" page appears and but won't go away and the shade doesn't close.
```

Nickduino, 2024-04-01:
> Does it mean the RX pin is not correct?

Possibly. IIRC, in the documentation and radio setup, **RX and TX refer to the CC1101**. So the RX pin is in fact the TX of your ESP32.

@rstrouse You could mention that in the wiring wiki and/or the Transceiver tab of the GUI.

cvhoang, 2024-04-01:
Ah after restarting ESP32, it's working properly. Thank you for this awesome software. Much better than RFXCOM hub it replaces.

closed 2024-04-01 by cvhoang