rstrouse/ESPSomfy-RTS #316: Letting the user choose the description of a group remote would avoid pagination issues

Nickduino, 2024-04-01:
[image]

There is another connected issue: a maximum of 8 shutters show up in the "link shade" section of the group tab. But it's fairly common to have more that 8 in a house **and** to require a general remote that opens and closes all shutters at once.

rstrouse, 2024-04-01:
Thanks for that it appears I have mis-calculated the line height for the group names and it should have truncated.   I also may just change this to instead of listing the names in the group to just report the number of shades in the group.   Now that rooms can be defined it really isn't as critical to call out each shade.

> There is another connected issue: a maximum of 8 shutters show up in the "link shade" section of the group tab.

On the list on the shades you have found a bug where the ESPSomfy RTS was truncating the list for the display.  This should support 32 potential shades per group.  I will have a fix in 2.4.2

Nickduino, 2024-04-01:
> Thanks for that it appears I have mis-calculated the line height for the group names and it should have truncated.

I **think** it did that with 2.4.0

>  I also may just change this to instead of listing the names in the group to just report the number of shades in the group. Now that rooms can be defined it really isn't as critical to call out each shade.

Again, I think the cleaner solution would be no description (the name of the remote should be descriptive) or custom description (which would be left blank by those you feel the name of the remote is descriptive enough)

rstrouse, 2024-04-01:
> I think it did that with 2.4.0

The addition of rooms in 2.3.1 pushed the panel height so it no longer truncates in the proper place and required some changes to the css.

> Again, I think the cleaner solution would be no description (the name of the remote should be descriptive) or custom description (which would be left blank by those you feel the name of the remote is descriptive enough)

I will try this out to see how it pans out.   Adding a custom string description is probably not in the cards given that I would then need to store an push it through the interface.  Now that you can put these in a room it is so much easier to identify them.

rstrouse, 2024-04-11:
@Nickduino I posted the 2.4.2 pre-release that now only indicates the number of members in the group.  This should be cleaner and doesn't require the type of CSS tricks needed to get the flex columns to act like I want.

rstrouse, 2024-04-21:
v2.4.2 has an official release please update to this release and you will see that only the member count now displays for groups.

closed 2024-04-21 by rstrouse