rstrouse/ESPSomfy-RTS #318: Same shades can be moved with one group but not with another one

eTx23, 2024-04-03:
### Hardware

ESP32

### Firmware version

v.2.4.1

### Application version

v.2.4.1

### What happened? What did you expect to happen?

I have a group called "all" of 17 shades to move all shades I have.
When I close all shades via the group "all" 3 shades stay opened. (I am sure those shades are part of the group because no more shades can be added to the group)
These 3 shades are also part of the group "living room". When I close the shades of the group "living room" also these shades close.

In this context:
The Home Tab show alle shades in the group "all". But when I edit the group "all" only 7 shades are shown.
One of the three mentioned above, which is not closing via the group control is shown in the group menu.

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-04-03:
I will have a fix for the shade list shortly but if some shades move and others don't on the group command it is because they do not have the group address registered in the motor.   In other words they are not members of the group.  The way this works is that ESPSomfy RTS transmits a single frame and any motors that  recognize the address respond to its movement.

Another user brought the fact that while the software allows you to add up to 32 members to the group, the display only shows the first 8.   I will try to get this fix posted this week.

rstrouse, 2024-04-03:
Unfortunately, you will have to wait for the fix to the list.  They are probably listed in the group on the server but I stupidly only display the first 8 so you cannot delete the ones that don't have the address stored in memory.

Nickduino, 2024-04-06:
> Just two questions:
> If those three shades are not member of the group why can't they be added to the group? If I want try to add a shade to the group no shade is available because all shades are already added to the group "all".
> Or are those not added properly?

They are "in the group" in ESPSomfy-RTS but the setup with the shutters probably didn't work as expected. So you have an information dichotomy:

- ESPSomfy-RTS thinks the group virtual remote controls those three shades;
- the shutters don't know that remote and won't move when they hear its orders.

My ESPSomfy-RTS doesn't work at the moment, so I can't check what I'm about to say... There is a tab in ESPSomfy-RTS that allows you to display a virtual remote and send commands by pressing the buttons. I hope (that's what I can't check) that this tab allows you to select a group remote (@rstrouse if it doesn't, it should) and to send the program command. If that's the case, you can put one of your three shades in learning mode by a long press on its physical remote, and then send the program command of the virtual group remote through the GUI. Hence solve your issue before the fix for the list.

rstrouse, 2024-04-11:
If you update to the pre-release of v2.4.2 of the firmware the shades list will show all of your shades and it will not overlap on the home screen because it now just shows the member count.

eTx23, 2024-04-15:
Hi,
thanks for the tip and the possibility to workaround the fix.
I tried to install the pre release and the firmware worked but the software didn't. I had to restore a backup, will try again and let you know.

closed 2024-04-16 by eTx23