rstrouse/ESPSomfy-RTS #319: Just asking some stuff

gralfj, 2024-04-05:
### Describe you new feature you'd like

first things first, Hi there

just reading the net I have found this project wich is very interesting to me.

I have several somfy shades, smome of them remote controlled, some by wall switchs

I would like to know about this, but seems this is not a forum like as far as I have seen.

Is there any place I can ask about my doubts

thanks in advance

Nickduino, 2024-04-07:
> of them remote controlled

Controllable by ESPSomfy-RTS **if** it uses the RTS protocole (i.e. not io homecontrol)

> some by wall switchs

Not controllable by ESPSomfy-RTS.

rstrouse, 2024-04-08:
To expand on @Nickduino 's response.  If your wall switches are Decoflex RTS or Smoove RTS they will work just fine.  However, if they say IO on them anywhere then they will not work.

Nickduino, 2024-04-09:
My bad, I understood "wall switch" as hardwired (so not radio based).

gralfj, 2024-04-09:
Hi, sorry, my little english hehehhe. By wall switch I meant hardwired, so
there is a wire from the switch to the blind.

rstrouse, 2024-04-10:
ESPSomfy RTS is capable of switching using relays. However, this requires a bit of electrical experience to wire them in.  See https://github.com/rstrouse/ESPSomfy-RTS/wiki/Controlling-Motors-with-GPIO

gralfj, 2024-04-11:
> ESPSomfy RTS is capable of switching using relays. However, this requires a bit of electrical experience to wire them in. See https://github.com/rstrouse/ESPSomfy-RTS/wiki/Controlling-Motors-with-GPIO

Hi, thank you for your reply.

I have read all the documentation on this github, 
what my little english understands is

- making a ESP device I would be able to control my blinds , the remote controlled and the hard wired ones,with a HA integration

 what I haven´t understood is

-  will the device need to be connected to AC electricity?
- how to setup hard wired motors and how many devices will be needed,

rstrouse, 2024-04-11:
The GPIO pins connect to a relay as a dry contact.  You will need to do some research online as most of these are 3 wire motors where there is 1 wire for common, 1 wire for up, and 1 wire for down.  Connecting a relay to the ESP32 is done via the pin triggers.

rstrouse, 2024-04-22:
I am going to close this one for now.  If you need further assistance, re-open this issue.

closed 2024-04-22 by rstrouse

gralfj, 2024-04-22:
Ok, thanks.