rstrouse/ESPSomfy-RTS #320: Unable to establish initial network ESP Somfy RTS

Findu28, 2024-04-05:
### Hardware

ESP32

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

I tried several different ESP32 (S3, C, ..), but none is able to bring up the initial network.
When i try the "Prepare for first use", all of them work fine.

### How to reproduce it (step by step)

```bash
1. Firmware programming - works fine.
2. Initial network - not shown
```

### Logs

```bash
...
```

rstrouse, 2024-04-07:
If the ESP32 is not booting into AP mode then please review whether you missed any steps from the wiki.  You must download the SomfyController.onboard.esp32xx.bin.zip file from the release and extract the file inside.   Then use that file to flash the board.

Findu28, 2024-04-07:
Thanks.
After several reprogrammings it now works.

rstrouse, 2024-04-08:
Awesome!  Glad you got it working.

closed 2024-04-08 by rstrouse