rstrouse/ESPSomfy-RTS #321: 500: Service error

Nickduino, 2024-04-06:
### Hardware

ESP32-C3

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

[image]

[image]

This is relatively common (something like once a week) but I don't know what triggers it. Are you aware of that issue?

### How to reproduce it (step by step)

```bash
1. Open Home Assistant, notice the shutter buttons aren't clickable;
2. reload the device in http://homeassistant.local:8123/config/integrations/integration/espsomfy_rts, notice it's stuck at "Initializing";
3. open the ESP32 own GUI and get the screenshots posted or an ERR_CONNECTION_TIMED_OUT (or a DNS_PROBE_FINISHED_NXDOMAIN if trying to use the mDNS)
```

rstrouse, 2024-04-07:
The mDNS for ESP32 is not great in that I don’t think it sends the appropriate packets for .local resolution when the network reserves the domain.   There could be something I am missing with it but it is supposed to be a couple of lines of code.

Essentially the gateway loses the reference.  Using the ip address bypasses the name resolution and adding an ip reservation on the router that includes the hostname also fixes it.

Nickduino, 2024-04-07:
I get the same errors when I connect to it with the IP address.

I mentioned the mDNS only because the message changes slightly when the connection to the device is completely impossible.

rstrouse, 2024-04-10:
I added some additional TCP publishing on the v2.4.2 pre-release.  Let's see where that takes us.

rstrouse, 2024-04-15:
@Nickduino have you seen an improvement with this issue after installing v2.4.2 pre-release?

rstrouse, 2024-04-22:
@Nickduino please make sure you update to the official v2.4.2 release.  There have been many improvements on the final release.

Nickduino, 2024-04-22:
Done. 

I haven't had any issue with the pre-release, though.

closed 2024-04-22 by Nickduino