rstrouse/ESPSomfy-RTS #322: Can see start grafik spinning wheel, but till login I gave up. After reboot blazingly fast login

fft1010, 2024-04-08:
### Hardware

ESP32

### Firmware version

v2.4.1

### Application version

2.4.1

### What happened? What did you expect to happen?

As in the subject: Since the 2.4.1 I this is the second time, that I had to reboot.
Last time I just rebootet (unplug/plug) today I gave up after abou 60 seconds watching the sinning wheel at the beginning.

I also have newst HomeAssistant.

### How to reproduce it (step by step)

```bash
Just wait :-) Sorry. No idea. I occurs. I am happily "just" using it. Occurs about every 2-3 days I do not know if it is related to the use we have with the shades. (More usage, stop ealier: I donno).
```

### Logs

```bash
in Home Assistant I see

2024-04-08 09:11:20.374 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback <_OneTimeListener <coroutine object start_server_discovery at 0x7ff56f4b5380>>(<Event homeassistant_start[L]>)
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/src/homeassistant/homeassistant/core.py", line 1322, in __call__
    self.hass.async_run_hass_job(self.listener_job, event)
  File "/usr/src/homeassistant/homeassistant/core.py", line 834, in async_run_hass_job
    if hassjob.job_type is HassJobType.Callback:
       ^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/backports/functools.py", line 68, in __get__
    val = self.func(instance)
          ^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 322, in job_type
    return self._job_type or get_hassjob_callable_job_type(self.target)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 354, in get_hassjob_callable_job_type
    raise ValueError("Coroutine not allowed to be passed to HassJob")
ValueError: Coroutine not allowed to be passed to HassJob
2024-04-08 09:11:20.377 WARNING (MainThread) [py.warnings] /usr/src/homeassistant/homeassistant/core.py:1753: RuntimeWarning: coroutine 'start_server_discovery' was never awaited
  return json_bytes(self._as_dict)

I do not know where I can find the ESPSomfy logfile on the ESP file system.

If you need anything else, just tell me.

thanks
Juergen
```

rstrouse, 2024-04-10:
The HA log above is not from the ESPSomfy RTS HA integration.   That is from another integration (most likely media player) that is trying to perform discovery.   

I have one other user that is having what appears to be a similar issue but it appears to be more frequent.  Which board are you using?  They are using a Wemos/Lolin board.

fft1010, 2024-04-12:
Hi @rstrouse,
I have no idea. See here the Foto, mayb it helps. If I should dig further I would need a hint. All I can say, that it started "out of a sudden", connected with the new update, but maybe not related.

[image]

Thank you
Juergen

rstrouse, 2024-04-15:
Try installing v2.4.2 pre-release to see if this alleviates your issues.  Some efficiency efforts have been applied that alleviate some of the pressure on the socket connection.  When the spinning wheel appears the webpage is trying to establish a socket to the ESPSomfy RTS webserver.

fft1010, 2024-04-15:
Hi @rstrouse,
I just did so. Anything I should keep an eye on?
thanks
Juergen

fft1010, 2024-04-15:
Hi,
I would post it here, if it "hangs" again. If you don't hear nothing, it works.
OK?
thanks
Juergen

rstrouse, 2024-04-15:
Yep I will wait a few days and close this issue if I don't hear back from you.

rstrouse, 2024-04-22:
@fft1010 I am closing this for now.  Please make sure you update to the released version v2.4.2

closed 2024-04-22 by rstrouse

fft1010, 2024-04-22:
Hi @rstrouse ,
closed 4 hours too early :-) This morning only reset helped again. We where away from last friday till yesterday (about 72 hours), where the system was not used at all. Yesterday evening it worked, this morning, I did not even see the spinning wheel; I did not get a connect to the server of RSTSomfy.
If you really think that it might be a hardware issue: What hardware do you use? If I exchange my hardware, we would be sure if it is hardware or not.

sorry
Juergen

rstrouse, 2024-04-22:
Before we go to hardware please update it to the released version of v2.4.2.   You had been running the pre-release and I fixed some issues related to the networking.

fft1010, 2024-04-22:
Hi @rstrouse,
I just did so.

thank you
Juergen