rstrouse/ESPSomfy-RTS #323: Shade not responding to group command

cvhoang, 2024-04-09:
### Hardware

ESP32-C3

### Firmware version

2.4.1

### Application version

2.4.1

### What happened? What did you expect to happen?

I have created a group with 4 shades ESPSompfy-RTS. Initially everything worked. However, one of shade is not responding to group commands. I've removed and relinked this shade but it still doesn't work. With Somfy Remote, I can still control all shades in the group.

### How to reproduce it (step by step)

```bash
1. Create a group with 4 shades
2. Send a group command
3. One shade doesn't respond
```

rstrouse, 2024-04-09:
You have likely inadvertently unpaired the motor from the group and when you are trying to pair it back in again it simply ignores you.  

You can pair it back in again by going to the group then pressing the delete button next to the linked shade.  When it prompts you to open the memory for the shade do not click on OPEN MEMORY click the NEXT button.  On the next screen press the UNPAIR FROM GROUP and respond that it jogged.

Next pair the motor back in again using the normal process.  Make sure that during the pairing process that you get the proper number of jogs.

cvhoang, 2024-04-09:
Thank you @rstrouse. I follow your instructions and the group commands are now working on all shades. However, it's pretty confusing how the unlink process works.

rstrouse, 2024-04-09:
Unfortunately, the motors never report the addresses that they are linked to and Somfy uses the same visual cues for linking and unlinking.  It needs to follow a similar process for each channel of a remote and you can create the same issue you created with the remote.  

During a normal link/unlink you would not be skipping any of the steps.  Oddly Somfy recommends unplugging all other motors that are not involved in the grouping process so you do not inadvertently open the memory of a motor that is not part of the group.  It is pretty easy to do if the wrong channel is selected on the remote.

I am going to close this for now.

closed 2024-04-09 by rstrouse