rstrouse/ESPSomfy-RTS #324: Boot Error

VictorLev1990, 2024-04-09:
### Hardware

Other

### Firmware version

2.3.0

### Application version

2.3.0

### What happened? What did you expect to happen?

After flashing few types of binaries in few types of controllers (WeAct esp32-c3, WROOM32 - same one shown in the "simple" wiki page),
Or the board get stuck, do not want to boot (ESP32-C3), or get stuck in a boot loop with the following message:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

I have tried to flash via Flash Tool, and ESPHOME, with the cc1101 connected and disconnected, always the same error.

Please advice.

### How to reproduce it (step by step)

```bash
*
```

### Logs

```bash
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57
```

rstrouse, 2024-04-09:
Invalid header means you did not flash (onboard) with a binary that is valid for your chip.  So if your board is an ESP32-C3 then you need to download the .zip file `SomfyController.onboard.esp32c3.bin.zip` but if this is a WROOM32 then you need to download `SomfyController.onboard.esp32.bin.zip`.   It cannot be both a WROOM32 and a C3

After you download the correct file specified above you must extract the zip to get the .bin file located inside.   Then use web.esphome.io to flash the file.  For more in-depth instructions see this wiki.
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Installing-the-Firmware

VictorLev1990, 2024-04-10:
My bad, downloaded the bin files instead of the ZIPs.
Thank you so much, great software.

rstrouse, 2024-04-10:
Glad you got it working!

closed 2024-04-10 by rstrouse