rstrouse/ESPSomfy-RTS #326: Saving as IOS web app. Only a few lines of code.

blaisebarrette, 2024-04-14:
### Describe you new feature you'd like

Hi all!

I love what you did! A breeze to setup. Worked first time!

I only have a little suggestion. When saving as a link on desktop of a IOS device. It opens a safari web page. With a few lines of code in the head of the page, when saved can become a web app.

A few lines of code that could make a big difference.

https://gist.github.com/tfausak/2222823

- Blaise

rstrouse, 2024-04-14:
Unfortunately, that requires each icon be pre-composed at its display size.   The icon sizes are too large to fit on the ESP 32 partition.  I will have to look for another way.  When I tried it without all the additional icons it simply did not load the socket.  I'll need to do some research to find out why it does this.

blaisebarrette, 2024-04-14:
Never mind the icons, although it would be nice but, really, the most important is this line:
`    <meta name="apple-mobile-web-app-capable"
          content="yes">
`
That gets rid of all the Safari interface.

rstrouse, 2024-04-14:
Yep I just figured that out.  I also figured out that it doesn't digest svg so scaling that way is not an option.  I am trying some compression to see what I can do for the icon.  Right now apple doesn't like a transparent background.

blaisebarrette, 2024-04-14:
This page seems like a good resource:
https://dev.to/oskarlarsson/designing-native-like-progressive-web-apps-for-ios-510o

rstrouse, 2024-04-14:
If you install the pre-release of v2.4.2 then the icons and the webapp settings have been applied.

blaisebarrette, 2024-04-14:
Wow! Niiiiiice!

rstrouse, 2024-04-15:
I am going to close this for now.  If you have any other issues please submit a new issue.

closed 2024-04-15 by rstrouse