rstrouse/ESPSomfy-RTS #328: Just spinning Wheel

xsasx, 2024-04-15:
### Hardware

ESP32

### Firmware version

2.4.0

### Application version

2.4.2

### What happened? What did you expect to happen?

Since I need the Shader again cause of the sun I found out that isn’t working anymore. The HA integration said it’s invailable- if I reload or read it seems to work for seconds - after that it’s unavailable again.

If I go to the webgui I see the grafical part of the gui an above it said radio not initialized and after minutes it said server not reaponding after X attemps. So the problem is I can’t check the config or do a fw update or backup :( what can I do ?

### How to reproduce it (step by step)

```bash
See above
```

rstrouse, 2024-04-15:
Remove power and let it reboot.  It looks like you somehow have a newer version of the application than the firmware.  Do you somehow have v2.4.0 of the firmware installed with the v2.4.2 application or is that the HA integration version.

rstrouse, 2024-04-15:
In a browser type http://ip address/backup where ip address is the ip address of the esp32.   This will backup your device.

rstrouse, 2024-04-15:
If you have internet connectivity and have backed up your configuration using the command above,  type the following command into a browser.

First run
`http://<ipaddress>/backup`

Then run.  This command will take a few minutes as it downloads the firmware and the browser may not return right away or even timeout.  Give it at least 5 minutes to do its thing.
`http://<ipaddress>/downloadFirmware?ver=v2.4.2

TibsD, 2024-04-23:
Hmm, seems like I have the same issue after updating to 2.4.2. I completed the github update process and after everything seems to be ok the browser did not reload. It just has a spinning wheel and an empty browser.

I tried the commands above but with no result...

TibsD, 2024-04-30:
> @TibsD did you get all sorted out?

Yes, I was able to upload a back up. Strange things was that I had to re-pair all motors because they did not respond. The ID did not change I just had to unpair them and go through the pairing process again. But everything works now.

rstrouse, 2024-04-30:
Awesome.  I am going to close this for now.

closed 2024-04-30 by rstrouse