rstrouse/ESPSomfy-RTS #329: Can´t add a shade to a group

AnadventureAlpacamybags, 2024-04-15:
### Hardware

ESP32

### Firmware version

2.4.1

### Application version

2.4.1

### What happened? What did you expect to happen?

Hello,

i got the problem, that i can´t add a shade to a group. If i press the OPEN MEMORY button for my selected shade, then nothing happens.  The motor dont jog, the cursor is rotating. After 6 seconds, the software ask me "did the shade jog"? 

No.

All shades work properly (up + down + My Position)

I'm missing something?

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-04-15:
Do you also have a remote?  If so using a channel that only controls the specific motor press and hold the prog button on the back of the remote until it jogs then press the Next button instead of the Open Memory button.  After that click the Pair to Group button and ensure it jogs.

If this does not work then you may have to factory reset the motor.  There are mixed reactions when the memory gets full on a motor.

AnadventureAlpacamybags, 2024-04-15:
Hi,

thank you for your advice to skip the OPEN MEMORY Button, it works fine. That was a easy solution.

rstrouse, 2024-04-15:
Awesome!  Glad I could help.

closed 2024-04-15 by rstrouse