rstrouse/ESPSomfy-RTS #330: I no longer receive from remotes since v2.4.2

blaisebarrette, 2024-04-15:
### Hardware

ESP32

### Firmware version

V2.4.2 pre

### Application version

V2.4.2 pre

### What happened? What did you expect to happen?

I now longer receive from remotes since I upgraded to v2.4.2 pre.
Neither in the radio logs or when creating a new shade an linking the remote.

### How to reproduce it (step by step)

```bash
I go to Radio->logs and I click on any remote buttons and nothing is logged.

It was working with v2.4.1
```

### Logs

```bash
No logs
```

rstrouse, 2024-04-15:
Refresh your browser

rstrouse, 2024-04-19:
@blaisebarrette did you get your issue solved?

blaisebarrette, 2024-04-21:
The shades are at my daughter’s home and I only had a few days to make this work. When I left, transmission worked well but reception not so much. I should should be visiting my daughter again end of may.

rstrouse, 2024-04-21:
Do some tuning and don't forget to update to the official v2.4.2 release.  There have been a lot of stability improvements.

rstrouse, 2024-06-30:
@blaisebarrette did you get your issues resolved?

rstrouse, 2024-07-06:
I am going to assume you got this resolved and close this for now.

closed 2024-07-06 by rstrouse