rstrouse/ESPSomfy-RTS #331: Kudos and some newcomer comments and questions

alka79, 2024-04-17:
Hello,
First let me give you KUDOS on the job done. What a project ! You set the bar very high :)
I recently moved to HomeAssistant and found your project that way. I wish I found it earlier.
It is polished in every aspect and fully documented. What a pleasure. 

I had it all setup in a few hours and it just works. I use v2.4.1 with a Wemos ESP32 S2 mini board and E07-M1101D transceiver board with it's small duck antenna.
The struggling part was the tuning of the radio. I never had any issue with any motor receiving from my remotes. The E07 transceiver at first did not receive properly one of my remotes, I could not figure out why this one. I tweaked the radio settings until it worked. I had to increase Rx bandwidth up to 117k. (no rational approach here, just try and test until it worked) 

Here are a few comments/questions in no particular order while it is still fresh in my memory:
- The doc mentions a few times EPS32S, but you really mean ESP32 S2. You had me questioning about the ESP32 chip range ;) The .bin to download is doubtless.
- This ESP32 board has usable pins nicely aligned, I was hoping for a simple cabling between the two boards and use two 4pin headers cables. Unfortunately, when trying to assign the Tx pin, the radio-buttons to select GPIOs 15 and 16 were missing. I can understand for pin15 as it is attached to the embedded led on this board, but why forbid the usage of 16 ? It is not a big deal as it is pretty easy to wire differently. 
- It seems that ESPSomfyRTS does not use the embedded led. Traditionally, the led blinks during setup and if wifi is lost. It gives visual feedback in case of trouble. 
- I sometimes open two web UIs, one on a pc and one on a smartphone. I noticed that the radio log was only updated in one of them, on the pc. Browser refresh did not help. Is that a limitation of the web server used ?
- By curiosity, in the suggested radio settings, is there a reason why you use weired numbers : 99.97 or 47.6. Why not round numbers? 
- On the System>Firmware tab, there is Firmware v2.4.1 and Application v2.4.1. What is the Application ? 
- HA integration works eventhough I did not set anything in the MQTT tab. That seems strange to me. Did I miss something ?
- I own plenty of old Somfy shutters that do not have the My function. I see the opportunity to bring that feature to them with ESPSomfy. I'll add my support to request #290 with some additional thinking. 
- I also miss voice control by Alexa. For this I will open a new topic to discuss what I have tried so far. 

Did I mention how great your project is ? thank you again for sharing such an awesome work with us
Al

alka79, 2024-04-17:
Thanks a lot. That clarifies. 

I'll look into HA Alexa integration and provide feedback, and will keep an eye on the 'virtual My" implementation :) Thanks again.

alka79, 2024-04-18:
Hi again,
About groups. The doc says:

> Groups do not have a defined position and cannot be controlled by setting the position. This is because the timings of each member can and will likely be different.

I created my first group today and noticed that when using the group commands, the position of each individual shutter in the group is properly updated.
This came as a nice surprise as I did not expect it from reading the doc text.
Does it mean you are close to allow setting a percentage target position to a Group ?

rstrouse, 2024-04-21:
> Does it mean you are close to allow setting a percentage target position to a Group ?

Yes at some point I plan on implementing a positioner on groups.  There are a few other things that need to be done first though.

closed 2024-04-21 by rstrouse