rstrouse/ESPSomfy-RTS #332: ESPSomfy with Alexa

alka79, 2024-04-17:
Hi,
My dream is to voice control my shades with Alexa from my chair ;)

Here is what I have done so far with HA and ESPSomfy RTS.

I know about 3 ways to add Alexa to HA. The advertised one is the subscription to Nabu Casa, another one is to get Amazon dev credentials and the last one, the least known, is what I used. 

HA has an embedded Emulated Hue bridge that can be used to expose entities to Alexa (and google assistant). This requires no additional integration nor credentials, just a few lines in configuration.yaml

```
# Emulated Hue for Alexa
emulated_hue:
  host_ip: 192.168.1.7          # HomeAssistant IP adress
  listen_port: 80               # port number 80 mandatory for Alexa and Google
  expose_by_default: false      # only entities listed below will be exposed
  exposed_domains:              
    - light
    - switch
    - cover

  entities:
    cover.shutter_1:            # entity ID of the shade
      name: "kitchen cover"     # name for Alexa
      hidden: false
```
Full restart of HA is needed to take into account. Then search for new Hue devices in Alexa.

The cover is recognized by Alexa as a Hue light, with the ability to turn On, Off and set brightness. Brightness from 0 to 100% translates to the cover position known by HA. For Alexa, 100% means "shade fully open", so you need to tick the invert position checkbox in ESPSomfy settings to be consistent.

Fortunately, Alexa also accepts the keywords Open, Close and set, far more natural for a shade.
You can say "Alexa, open the kitchen cover", "Alexa, close kitchen cover", "Alexa, set kitchen cover at 30%" or "Alexa, lower kitchen cover by 20%". This all works.
You can even see the cover brightness (position) slider updated live in Alexa UI on your smartphone while the shutter moves.

Although this works like a charm for Open and Close voice commands, there is a glitch when setting the position via voice. After a few tries, Alexa starts complaining. When something changes on the device, the device reports it's status back to Alexa, including position. If there is a discrepancy between the setting asked by Alexa and the feedback received, Alexa concludes that the device does not react properly and complains like "the device seems malfunctioning blah blah". From then on, she refuses to set a position by voice until the device status is reset correctly, for example when fully On or Off. 
To explain, first, Alexa is not used for a light to take several seconds to reach an assigned level :  there are probably timer issues. Second, Alexa is very sensitive. I noticed that in a previous project. If you ask Alexa to set to 50% and, due to rounding errors, she receives a feedback of 49% or 51% from the device, she will complain.

Setting cover position by voice is a must have and Alexa complains to frequently to consider this simple solution as acceptable.

Any thoughts welcome
Al

rstrouse, 2024-04-17:
I'm not sure there is any way to fool the Hue skill into thinking that the ramp time = up/down time.   I am guessing this is most likely a timing issue since ESPSomfy RTS sets the integer position at the end of the movement and eliminates any potential rounding issues from the target sent.

I am pretty sure, although I do not use it, that the Alexa skill in HA works just fine.  I would expect that the embedded VA works fine as well.

alka79, 2024-04-23:
I close this one as it is related to HA and should be addressed in the HA community forum.

closed 2024-04-23 by alka79