rstrouse/ESPSomfy-RTS #334: local build issues

alka79, 2024-04-19:
Hi,
Initially I installed the contoller via ESPHome web. That worked like a charm.

I downloaded 2.4.1 source code to build it locally. I changed FW_VERSION to 2.4.1a to notice a difference. 

I could compile and upload the newly built SomfyController.ino.esp32s2.bin with the web UI. I had to manually reboot the board at the end though. The newly uploaded firmware was running with v2.4.1a.

However after using the web ui for a short while I started getting Service Errors 500. 
GET shade?shadeID=12 when trying to access that shade in the settings or /getsecurity randomly with endless spinning wheel.
I had to manually reset the board and it worked again... for a while.

I upload twice my bin file with the same results.
Since it failed, I re-uploaded via the app the official 2.4.1 bin file, and now it is back to normal.

I must be doing something wrong or use wrong settings. 
I use Arduino 1.8.19 on windows 10, latest versions of all libraries with an ESP32 S2 mini board.
My bin file is 1 199 520 bytes. The esp32s2 you built is 1 169 104 bytes.
Here are my options in the Arduino tool menu.

[image]

Unfortunately with this board, Arduino does not connect to the serial port to see the messages printed there. 

Any hint ?

rstrouse, 2024-04-19:
Likely your issues where the endpoint can't be found are an mDNS issues.   Try using the ip address in your GETs.  Also, be sure you are compiling with:
* Arduino Core  v2.0.14
* Websockets v2.4.0
* PubSubClient v2.8.0
* ArduinoJson v6.21.5

It is very odd that you are not seeing the serial port when CDC on boot is enabled.

alka79, 2024-04-19:
I use the IP address not the hostname.
CDC on boot seems to work only if arduino is used to upload, at least it worked with some basic sample sketches I tried earlier. 
I might change for a wroover board if it is to painful with this one.

What about the Partition scheme ? I left it to the default

I run on slightly newer versions of the libs. Will try with the same as yours.

rstrouse, 2024-04-20:
ArduinoJson v7+ eats up too much codespace at this point and core 3 is only in alpha.  Are you using export compiled library?  That should compile it with the proper CDC settings for the serial port.

rstrouse, 2024-04-21:
The official release of v2.4.2 fixes some UDP leakage that would cause the unit to crash after several days.  Please update with that code.

alka79, 2024-04-22:
I updated today to 2.4.2 and issues with undefined groups are gone. 

The update process was a bit hectic. 
- Firmware update by uploading my own build (with USB CDC on boot enabled) was no brainer. I could see the 2.4.2 Serial output right after. 
- App update by uploading your Littlefs.bin file failed at the end of upload and from then on, no web UI anymore. Arduino "ESP32 LittleFS data upload" tool not working with an ESP32-S2 board.

I had to reinstall from the start with ESP Home web.
Thanks to restore of the backup file, I was quickly back on track :)

rstrouse, 2024-04-22:
I noticed in the Arduino core that there was an update fix for the OTA where it would intermittently fail if there was a pause in the byte stream.  It only happens under very specific conditions but I suspect that is what I have seen from users a couple of times.

I will incorporate these changes in the next release.

closed 2024-04-23 by alka79