rstrouse/ESPSomfy-RTS #335: backup file has wrong format

rgr101, 2024-04-19:
### Hardware

ESP32

### Firmware version

v2.4.2

### Application version

v2.4.2

### What happened? What did you expect to happen?

Something went wrong when I updated to the latest 2.4.2 pre-release (probably my own fault, as I was in a hurry and may have made a mistake). Luckily I had done a backup right before. 
I had to re-onboard the ESP and wanted to do a restore. However, the backup file had the wrong format: the extension was ".txt". 
Maybe it was because I made the backup with the iPhone (iOS 17.4.1) and once again an unintentional "Apple Magic" hit me (normally I love the Apple Magic, but sometimes it's just annoying...)? 
Fortunately, I was able to solve the problem by simply renaming the file and removing the extension. 
Now that the ESP is up and running again, I made a backup and noticed that the extension .txt is actually suggested in the file dialog. 
I had overseen this and didn't know it would cause problems anyway...

By the way, the restore worked perfectly! 👍👏👏👏

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-04-20:
Unfortunately, I cannot control what shows up in the dialog since that is based on the mime type for the file.  Most of the fruity magic I could do without.   Anyhow, I will allow the backup to accept .backup and .backup.txt.

rgr101, 2024-04-20:
👍☺️🙏🙋‍♂️

closed 2024-04-20 by rgr101