rstrouse/ESPSomfy-RTS #336: Two remotes, one is seen the other is not.

jurgen2005, 2024-04-19:
Hi I have 1 awning and two remotes.
Both remotes operate the awning perfectly.
ESPsomfy operates the awning perfectly.
Only one of the remotes is seen by ESPsomfy, the other one is not, no matter what I do or how I scan.

please advice, regards Jurgen

rstrouse, 2024-04-20:
The remote that does not work may be a little bit out of tune,  Slowly increase the RX bandwidth and press save in between until you get a signal from the remote.

jurgen2005, 2024-04-20:
hmm its either one controller or the other when I move the RX. 
Could it be they are that far apart, while the shade responds to both?

rstrouse, 2024-04-21:
Unfortunately, the tolerances on the CC1101 appear to be a bit tighter than the receiver on the motor.   Also reduce your frequency deviation to see if it picks them both up.  You will find that the Somfy remotes really do not transmit at very high power.  Reducing the deviation will make the CC1101 interpret weaker signals if you do not have a whole lot of interference.

rstrouse, 2024-04-22:
I just wanted to say one more thing about this.  I have had a couple of E07-M1101 transceivers that were a little weak.  I don't know if they just have a manufacturing defect or if there was some component that was goofy on them.  One looked like there was a dent in the crystal.

rstrouse, 2024-05-11:
@jurgen2005 did you come to any conclusions on this?

jurgen2005, 2024-05-13:
Hi, no I gave up :) its not that big of a problem.

rstrouse, 2024-05-13:
I'm not sure how I would accommodate that in the firmware.  You may be able to get something like that done in HA where you issue a stop when it reaches a certain percentage and it is opening.

rstrouse, 2024-05-21:
I am going to close this one for now.

closed 2024-05-21 by rstrouse