rstrouse/ESPSomfy-RTS #337: Cannot Pair the blinds

SteveButt, 2024-04-20:
### Hardware

ESP32

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

After factory reset I cannot pair any blinds

### How to reproduce it (step by step)

```bash
1. Paired three sets of blinds but copied the remote ID and used the same ID in the App
2. It worked but the control and App get out of sync
3. Remove all blinds from the App and Factory reset all blind motors
4. Try to re-add he motors with different ID numbers ( Unique now )
5. Motors never pair, do not jog when pressing or holding Pair button in App
```

### Logs

```bash
[{"encKey": 164,
  "address": 10165047,
  "rcode": 423,
  "command": "Down",
  "rssi": -61,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [73012,23004,6396,62296,8302,25417,6388,68201,27806,6383,51498,14325,36758,4935,58295,32599,6389,61016,41376,19384,10495,7280,2507,2527,2513,2522,4843,1280,1320,1267,1320,638,636,1296,1304,625,669,1279,657,645,645,634,1310,1270,660,640,649,644,1303,1275,649,637,660,647,1294,1292,644,649,658,641,655,615,1301,1291,1294,639,653,1292,1312,620,669,634,660,1275,648,637,1309,1295,1286,642,648,1293,644,644,655,639,656,645,1295,640,642,1300,1291,648,657,640,652,1270,659,640,652,643,652,639,659,619,670,634,662,625,644,650],
  "time": "2024-04-20T16:42:10.770+0100"
},
{"encKey": 165,
  "address": 10165047,
  "rcode": 424,
  "command": "Down",
  "rssi": -62,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [95993,6401,1493,12835,59279,465,3547,18349,6423,6151,10476,7280,2520,2545,2498,2519,4856,1285,1292,1282,1304,650,659,1267,1320,1261,664,649,659,635,650,618,1328,627,664,626,659,640,650,1271,661,641,649,644,655,641,1307,621,656,641,655,644,646,636,661,648,643,1280,1314,628,660,1293,1293,639,659,618,669,1288,651,642,1304,1276,1298,642,648,1294,662,640,645,623,657,641,1304,649,657,1270,1319,616,658,646,643,1283,677,614,676,638,659,613,668,629,661,631,672,625,653,639],
  "time": "2024-04-20T16:42:12.360+0100"
},
{"encKey": 165,
  "address": 10165047,
  "rcode": 424,
  "command": "Down",
  "rssi": -60,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [630,9423,15338,504,1706,2547,2545,2502,2530,2518,2511,2521,2542,2508,2528,2514,2548,2484,2548,4855,1268,1311,1272,1308,641,645,1301,1285,1296,636,640,648,645,656,639,1305,647,632,641,653,642,675,1259,666,647,658,614,676,616,1308,640,667,635,638,639,674,621,656,634,655,1295,1310,620,669,1285,1301,627,660,641,651,1293,635,641,1299,1284,1303,650,633,1291,668,628,660,639,649,645,1304,624,666,1280,1290,650,659,638,644,1298,640,642,648,637,660,647,644,656,636,640,648,645,657,638],
  "time": "2024-04-20T16:42:12.483+0100"
},
{"encKey": 165,
  "address": 10165047,
  "rcode": 424,
  "command": "Down",
  "rssi": -61,
[... 475 lines cut ...]
{"encKey": 163,
  "address": 10427191,
  "rcode": 91,
  "command": "Prog",
  "rssi": -62,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [7800,8498,565,10714,2523,2518,2502,2543,2499,2544,2514,2525,2515,2526,2512,2518,2515,2532,4850,1270,1312,1270,1314,619,670,634,635,1301,647,637,1311,617,676,1285,1292,636,664,1276,1298,641,649,643,653,634,665,1269,1311,644,646,1285,1301,628,659,632,649,1297,662,626,649,646,658,641,651,641,652,638,634,646,643,658,634,640,673,619,1309,648,643,630,659,639,665,628,659,1267,667,630,661,641,650,643,653,638,1308,1270,660,624,666,629,660,641,1300,629,659,1282,644,658,1287,643,647],
  "time": "2024-04-20T16:54:21.264+0100"
},
{"encKey": 163,
  "address": 10427191,
  "rcode": 91,
  "command": "Prog",
  "rssi": -64,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [6609,5086,15868,2522,2539,2497,2542,2506,2527,2517,2519,2513,2528,2513,2522,2506,2528,4857,1270,1311,1271,1308,639,646,649,633,1292,669,627,1312,620,669,1287,1279,649,659,1267,1319,638,633,645,670,632,661,1264,1296,638,653,1295,1310,620,668,634,660,1275,647,637,658,648,643,656,636,641,648,644,654,640,656,643,645,636,660,625,1319,638,634,647,645,657,636,641,649,1294,660,640,645,647,633,641,654,644,1300,1291,643,631,661,629,671,646,1283,649,641,1283,654,644,1299,630,672],
  "time": "2024-04-20T16:54:21.405+0100"
}]
```

SteveButt, 2024-04-20:
These are brand new genuine Somfy Roller blinds motors with a Situo Pure II 5 channel remote
My hardware is exactlt as reccomended on the README i.e. ESP32 with E07-M1101D-SMA wired as per the README.
The ESP/RADIO unit is no more than 6 or 7 feet from any blind.

rstrouse, 2024-04-22:
Hmm.  So your ESPSomfy RTS device has been transmitting in the past? Is that correct?  Just to be sure when you long press on the prog button on the remote do you get a jog?

SteveButt, 2024-04-22:
Hi, Yes, it has been transmitting in the past and yes a long press on the physical remote results in a Jog, just nothing when I press the "Pair Shade" tried pressing that bot short repeatedly and long in case it helps but I never get a jog.

rstrouse, 2024-04-22:
Which motors are these?   Some motors require the endstop limits to be set prior to pairing any other remotes on initial install and after a factory reset.

SteveButt, 2024-04-22:
Not Exactly sure, They are listed as 0.7nm Liion battery motors, they are roll up blind motors if that helps narrow it down. I could take the blind down and try to find the label if needed but I did set all the endstops with the remote prior to trying to pair them.

rstrouse, 2024-04-22:
Did these say Somfy in big letters on the tube?

SteveButt, 2024-04-23:
Yes - This is the first unit https://www.somfypro.com/products/-/e-cat/1240278/Roll_Up_24_WireFree_RTS_Li-ion
On The Tube it says SOMFY on one side, on the other it says 1240278B ROLL UP 24 WF RTS LI-ION

rstrouse, 2024-04-23:
In the instructions the orange led on the motor should be blinking when the memory is open.   Then the green led on the motor should blink 5 times when ESPSomfy RTS is recognized.  Are you seeing the orange led blinking?

SteveButt, 2024-04-23:
The Orange Led does blink but it says in that state, when I press the "prog" button in ESPSomfy, nothing changes, after a few minutes the blind jogs and the orange light stops blinking.

rstrouse, 2024-04-23:
Try changing the address to something else on the shade definition and try again.  Also, please double check your wiring to make sure nothing came disconnected especially with GDO0 - TX pin or changed from when it was sending in the past.  This should just be working.

SteveButt, 2024-04-23:
Changed the address, but exactly the same behaviour, the new address will not clash with anthing from the remote as I notice all the addresses the radio log sees from the remote are 8 digit, the default and updated address on ESPSomfy blind definition are 6 digit. I Checked the wiring, everything firmly in place, nothing hardware wise has changed unless somehow the radio transmitter has failed but that wouls sem like an awfully big co-incidence. Howeverr, if I go back to using the remote's address then it still doesnt work so maybe that what it is ?

rstrouse, 2024-04-23:
Give it a shot.  It is sending the proper frames.  I assume you held down the prog button on ESPSomfy RST when you were trying pair.  Reading the instructions there are multiple cases that when pressing the button with the paper clip that it blinks the leds.  Are you sure the process you are using is correct.  Somfy does this a lot where the same response means different things depending on what you did to get it to jog.  For instance, pairing and unpairing are the same process.

SteveButt, 2024-04-23:
There are many mentions of prog buttons etc so perhaps I am misundrstanding. When we get  to adding the blind it ( I think ) refers to pressing the prog button on the back of the physical remote, then to pair I press te button on screen on the ESPSomfy interface ?

rstrouse, 2024-04-23:
Yes if you press and hold that button in ESPSomfy RTS it will send a longer frame series.  There is also a pairing scenario where you use the prog button on the motor head and a paper clip.  The doc also refers to a remote as a control point.

SteveButt, 2024-05-01:
I could only find the prog button option on the motor for straight after a reset. I did a reset and re-paired with remote but the same.
I bought another radio module , same, I wired it the DSUN way when it didnt pair but then I dont see anything in the Radio logs. I re did the wiring a couple of times, nothing. I have ordered another ESP32 board as a last option, will try tomorrow.

rstrouse, 2024-05-01:
If you do not have a DSUN module then it should not be wired that way.  It will say DSUN on it and most likely be green instead of blue.   The prog output you have from the remote is identical to the output that is being sent from ESPSomfy RTS so I am not sure why it is not pairing.  Is the shade definition set to RTS 56-Bit?

SteveButt, 2024-05-01:
The DSUN wiring was a long shot for sure, and bothe radio modules look identical to your pictures. The shade definition is set to 56bit.

closed 2024-05-02 by SteveButt

alextrical, 2024-07-21:
I had the exact same issue, however the solution was re-assigning the RX and TX to GPIO16/17 (RX2/TX2) pins.

It seems like there is an issue when using GPIO12/13 on some clone ESP32 DEV KIT v1 modules