rstrouse/ESPSomfy-RTS #338: ESP32: Unable to reconnect to wifi

chemelli74, 2024-04-21:
### Hardware

ESP32

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

When I restart my router or there is a power cut, then the device is not able to reconnect to wifi.

### How to reproduce it (step by step)

```bash
1. power cut
2. wait for the router to restart and offer wifi services
3. verify that the device doesn't reconnect
```

rstrouse, 2024-04-21:
Is the AP available when this occurs?

chemelli74, 2024-04-21:
> Is the AP available when this occurs?

Need to check

rstrouse, 2024-05-11:
@chemelli74 I did some work to better recover from a condition where there is a power cut and the router is not ready to accept connections yet in v2.4.3.  In some cases it was possible to get the SSID to appear and start the connection but still never get an IP address from the network.   In essence it would indicate it was connected but without any registered IP address.

I have detected this condition and bumped the Arduino core which fixes the premature connection reporting.  v2.4.3 will be going to general release this weekend.

chemelli74, 2024-05-11:
Thx, will update and report back at the first occurrence

rstrouse, 2024-05-12:
I am going to close this for now as I do believe I replicated your issue and found where it was being reported as connected although there was no IP address.

closed 2024-05-12 by rstrouse