rstrouse/ESPSomfy-RTS #339: groups displayed as Undefined

alka79, 2024-04-21:
### Hardware

ESP32-S2

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

I created 4 groups. The first was created some days ago without any issue, 3 mores today.  
During the process of adding groups today when navigating between tabs, some groups randomly become Undefined. In both the Home tab and in Somfy>Groups.
Refresh the bowser may not hange anything or may display other groups as undefined.
The backup file is correct though. When I reboot the board they are all back to normal... for a while. 
It seems to be a display issue in the app.

Some examples taken in less than 10 minutes time:
[image]

[image]

[image]

[image]

### How to reproduce it (step by step)

```bash
randomly appears when navigating between tabs or refreshing the tabs
```

rstrouse, 2024-04-21:
I will have a fix for this soon.  The issue is related to the json serialization libraries.

rstrouse, 2024-04-21:
Update to the v2.4.2 release and you problem should be solved.

rstrouse, 2024-04-22:
I am going to close this for now as I believe we got you sorted.

closed 2024-04-22 by rstrouse