rstrouse/ESPSomfy-RTS #340: Unable to find any signal when scanning frequencies

Chaospilot1980, 2024-04-21:
### Hardware

ESP32

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

I got everything running, initailizing the radio is done. But i cant find any signals at all. tried 4 different remotes, but the radio does not pcik up anything. not in logfilæes, not upon scan frequencies.

Wiring is:
Radio Module - color - ESP32 board
1/GND - Grey - GND
2/VCC - Purple - 3,3
3/GDO0TX - Blue - 13
4/CSN - Green - 05
5/SCLK - Yellow - 18
6/MOSI - Orange - 23
7/MISO - Red - 19
8/GDO2RX - Brown - 12

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-04-21:
Ok I have determined that your wiring is correct if your radio pin settings look like this.
[image]

Just to be sure your remotes do not say IO on them anywhere when your read the back of it.   The led on RTS remotes will most often be red in color when you press a button on the remote.  The radio for IO HomeControl is fundamentally different and will not work with this solution.  This only works with RTS.

If that is the case verify that you also have 433.42mhz selected.

rstrouse, 2024-05-09:
@Chaospilot1980 did you get your issue resolved?

rstrouse, 2024-05-21:
@Chaospilot1980 I am going to close this for now.  Feel free to reopen if you want to continue to troubleshoot.

closed 2024-05-21 by rstrouse