rstrouse/ESPSomfy-RTS #341: Can't pair one of my shutters

NeatMonster, 2024-04-22:
### Hardware

ESP32

### Firmware version

v2.4.1

### Application version

v2.4.1

### What happened? What did you expect to happen?

Hello,

First of all, thank you for this great project!

I have been using ESPSomfyRTS successfully to control RTS-enabled shutters and awnings. But I have recently installed a Somfy's [RTS Micro-receiver](https://www.somfy.co.uk/products/2401162/micro-receiver-rts-for-shutter) for a wired shutter and can't get ESPSomfyRTS to pair to it.

I've had to increment the RX Bandwidth value to get ESPSomfyRTS to recognize the remote. However,  I cannot get the pairing with the micro-receiver to work. Pressing the `Prog` button for 3s on the micro-receiver (or on the remote I've paired with it) makes the motor jog, but then nothing happens when pressing the `Prog` button on ESPSomfyRTS' interface.

The ESP32 and micro-receiver in the same room and very close, so I don't think this could be a radio signal strength issue. I will paste below a log of the radio commands from the paired remote to the micro-receiver.

Any idea of what might be going wrong? Or what I could be doing wrong?

Thank you for your help!

---

[image]
[image]
[image]

### How to reproduce it (step by step)

```bash
1. Go to Somfy > Shades > Edit
2. Click on Pair Shade
3. Press the Prog button for 3s on the micro-receiver/remote
4. The motor jogs
4. Press Prog on the interface
5. Nothing happens
6. Try pressing Prog again (for longer) on the interface
7. Nothing happens
```

### Logs

```bash
...
{"encKey": 174,
  "address": 15279186,
  "rcode": 663,
  "command": "Up",
  "rssi": -70,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [10218,18408,2506,2523,2475,2518,4799,1271,1286,1290,1298,1267,644,647,633,640,1280,1295,1287,619,669,608,664,642,649,618,650,1289,1275,1274,1311,618,648,637,661,627,644,622,659,640,658,621,643,631,660,632,639,645,654,1262,1291,1283,637,643,650,635,635,641,646,644,654,615,658,623,666,628,661,633,649,617,1301,1267,645,632,1313,1273,653,614,1286,1295,1283,1265,1302,641,649,618,652,1282,1292,628,661,616,654],
  "time": "2024-04-20T20:04:59.389+0200"
},
{"encKey": 175,
  "address": 15279186,
  "rcode": 664,
  "command": "My",
  "rssi": -68,
  "bits": 56,
  "proto": 0,
[... 246 lines cut ...]
  "address": 15279186,
  "rcode": 667,
  "command": "Down",
  "rssi": -56,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [10237,18417,2512,2526,2460,2529,4808,1279,1282,1268,1286,647,644,629,661,1266,1293,1289,624,644,650,635,1286,1295,1284,615,646,645,656,1265,645,644,655,637,1282,1268,1287,1273,1302,640,648,1268,660,633,640,643,1279,641,648,643,628,1287,648,643,1278,640,650,1270,635,666,1276,646,633,639,631,1297,1279,1285,641,654,635,641,626,644,1301,613,652,1294,634,639,656,620,1296,1265,647,644,657,638,1283],
  "time": "2024-04-20T20:05:27.110+0200"
},
{"encKey": 163,
  "address": 15279186,
  "rcode": 668,
  "command": "My",
  "rssi": -58,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [10231,18425,2483,2527,2481,2509,4804,1277,1284,1296,1286,650,618,650,635,1292,643,646,634,640,1276,1295,635,640,647,642,1279,1291,1269,1288,1275,1302,633,625,644,650,1286,649,643,626,635,641,653,644,648,1286,1276,646,633,1292,1272,631,659,1283,643,628,1287,649,643,1279,630,647,1295,1262,645,643,655,636,640,643,1274,1284,1294,635,633,1290,645,629,1288,1298,627,636,660,625,1296,1284,626,643],
  "time": "2024-04-20T20:05:28.759+0200"
},
...
```

rstrouse, 2024-04-22:
I do not see anything weird in the pulse outputs from what I assume is the remote.  ESPSomfy RTS should be sending the exact same symbols.   Lets try this.  Change the configuration address and try again after removing power from the micro receiver for a few seconds.

If that doesn't work factory reset the micro receiver and start over.

rstrouse, 2024-04-22:
Also did you hold the PROG button down in ESPSomfyRTS?

NeatMonster, 2024-04-23:
> I do not see anything weird in the pulse outputs from what I assume is the remote. ESPSomfy RTS should be sending the exact same symbols. Lets try this. Change the configuration address and try again after removing power from the micro receiver for a few seconds.

That did the trick, thank you! I have tried it before opening this bug report, sorry.

closed 2024-04-23 by NeatMonster