rstrouse/ESPSomfy-RTS #342: Somfy blinds initially non responsive

mozzaLB, 2024-04-22:
### Hardware

ESP32

### Firmware version

V2.4.1

### Application version

V2.4.1

### What happened? What did you expect to happen?

No problems pairing multiple Somfy motors, but the response from the user interface is unreliable (there's no issue using the remote). Initially, the web-ui is non-responsive to any of the buttons. If I click one of the buttons multiple times, it suddenly responds, and after that, it responds immediately to any button press. I encounter the same issue with all blinds. It seems that the motors need to wake up initially. What could be the problem?

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-04-22:
Please update to v2.4.2.  There have been some updates to the network handling.  

However, this could also mean that the mDNS name are not being kept by your network.  So what happens is the web interface asks for the local address and the router has to ask all the stuff connected if it belongs to them.  Try using the IP address to troubleshoot this.  If it works consistently then we need to look at how the local mDNS is setup on your network.   Also if you are using .local try it without anything just the hostname.

Another thing that commonly is required on your router is to set up a DHCP reservation that includes the host name of the ESPSomfy RTS device that you set in the Options tab.

mozzaLB, 2024-04-22:
Thanks for your quick response. The IP address is already reserved for the ESPSomfy RTS device in my router and is also set in the option tab (Acquire IP Address automatically DHCP is selected). I have updated to V2.4.2, but the issue remains the same. Also no difference with the IP or the Espsomfyrts.local address. If I triple-click a random button, then every button works immediately afterwards. However after a few minutes, I have to triple-click again. This happens with all blinds.
Initially I did not have this problem, but I recently installed 2 additional blinds and updated from 1.7.2 to 2.4.1. Perhaps I could revert back to an older version to test if this solves this issue?

rstrouse, 2024-04-22:
If you are running a CC1101 12db will send out intermittent garbage.   Only under specific chips can you reach 12dB gain set it to 10 for the E07-M1101 module.   Open the browser console and see if there is an error in there when first pressing the button.  If there isn't increase the number of repeats in the shade definition.   Some motors go to sleep to conserve energy and it is near impossible to get a single frame sent from a handheld remote.

mozzaLB, 2024-04-22:
The number of repeat commands in the shade definition did the trick, how could I have missed that option. I have set it to 3 times and now it works like a charm. Thanks a lot for your quick help with this! There was no error after the first press in the browser console btw

rstrouse, 2024-04-22:
Awesome.  I am going to close this for now.

closed 2024-04-22 by rstrouse