rstrouse/ESPSomfy-RTS #343: Multiple shutters control at the same time gives problems

Electriceye1091, 2024-04-22:
### Hardware

ESP32

### Firmware version

v2.4.2

### Application version

v2.4.2

### What happened? What did you expect to happen?

When controlling multiple shutters in HomeAssistant at the same time to a position other than open or closed they all do something different.
One always does the same, which is nothing.
The other two alternate on going to the correct position and opening all the way. (It's always one opening completely and the other to the correct position).
When using the open, close and stop button in HomeAssistant they all act like they should as a group and individually.
Haven't tried a fixed position via the webinterface because I only see the standard buttons.

### How to reproduce it (step by step)

```bash
When using the slider in HomeAssistant to make a group go to e.g. 60% they all do something different.
When using the slider on individual shutters the all act accordingly.
```

rstrouse, 2024-04-22:
Upload a backup of your setup.   It is starting to look like I need to do some work on the standard HA cover group.   On my test rigs it all works as it should.  But that doesn't mean I covered all the scenarios.   The reason why the web app does not include a positioner is because the position is non-deterministic.  Especially when you add awnings or blinds to the group.

Also if you installed the pre-release of v2.4.2 please install the official release.

Electriceye1091, 2024-04-22:
For some clarification. I made the group in the webinterface
[image]

rstrouse, 2024-04-22:
That is very odd.  I tried this in HA and it worked as expected.   I see the down commands issued as well as the stop/my commands issued.  Is the ESPSomfy RTS UI correct?  If it is add a repeat to each of the shade definitions.

Electriceye1091, 2024-04-23:
The web interface is showing what position the shutters are supposed to be. I will try the repeat after I come home from work.

Electriceye1091, 2024-04-28:
Sorry for the late response. It's been a little busier than anticipated. Adding two repeats seem to have fixed the control from home assistant. I will try to tune it for the control via home assistant via Apple HomeKit. That's still a bit buggy but that's probable the communication between crApple and homeassistant.

rstrouse, 2024-04-29:
Awesome I am going to close this for now.  Most likely the motor was too slow after it decoded the message targeting another motor.  The repeats give it an opportunity to reset between these commands.

closed 2024-04-29 by rstrouse