rstrouse/ESPSomfy-RTS #344: Comments and question (newcomer in progress)

alka79, 2024-04-23:
While I am still discovering,
- repeat is an interesting word. In the settings, 'Repeats commands' is empty by default, which is same as 1. If we want to repeat the commands one more time for a shade, we must enter 2 here. Correct ?

- memory usage. WebUI requires a lot of RAM. The number of web clients is not limited by the system, but in practice, with two webUI clients the heap is already very low. During my current test and setup phase, I frequently had the app running on my pc and my smartphone at the same time.  I guess that explains some malfunctioning experiences. It should be mentioned somewhere that one webUI client is ok, two at the same time becomes risky.

- Serial prints non stop. This morning I found the Serial messages window full of "Timing Somfy: xxxms" with xxx taking various values from 139 to 300ms. It was going on and on,  every second. The board was idle. Something I should be worried about ? I reset the board and it came back to normal.

- suggestion. It's great to see the min heap in 2.4.2. Maybe you could add a time stamp to better track. Also display the last reset and last wifi reconnection timestamps. It helps to debug. For example, I had a situation when the app was not reacting anymore. When looking at the serial output, i could see that I had a Wifi connection loss. I just had to wait a few seconds until it reconnected.
Why the board loses WiFi from time to time is another concern that I need to further investigate.

- appversion file is not update to 2.4.2

rstrouse, 2024-04-23:
> repeat is an interesting word. In the settings, 'Repeats commands' is empty by default, which is same as 1. If we want to repeat the commands one more time for a shade, we must enter 2 here. Correct ?

ESPSomfy RTS will always send it once.  If you want two commands you would have 1 repeat.   Bear in mind a repeat is the repeating of the initial frame but it contains a different start of the frame for the repeat.

> memory usage. WebUI requires a lot of RAM. The number of web clients is not limited by the system, but in practice, with two webUI clients the heap is already very low. During my current test and [...]

That is not really the case.  I typically have 4 open for 3 of my 10 test devices an see only about 2k per connection for the client buffers However, the underlying idf will drop sockets after 5 are opened depending on the board.  I think the ESP32 maxes out at 10 but the C series chips limit this to 5.

> Serial prints non stop. This morning I found the Serial messages window full of "Timing Somfy: xxxms" with xxx taking various values from 139 to 300ms. It was going on and on, every second. The board was idle. Something I should be worried about ? I reset the board and it came back to normal.

I don't typically see this behavior at idle occasional instance where the processing time increases when it has to decode a bunch of frames or send a bunch.   I would not worry about it the timings are in there so that I can keep the response chain low.

> suggestion. It's great to see the min heap in 2.4.2. Maybe you could add a time stamp to better track. Also display the last reset and last wifi reconnection timestamps. It helps to debug. For [...]
Why the board loses WiFi from time to time is another concern that I need to further investigate.

I am going to leave the logging and timing functions to the automation package and the HA plugin will be getting an update.   There is no RTC on the ESP32 so time is an illusion created by the NTP service.  When you start getting into storing timespans you shoot into the 64bit arena after 45 days.

Keep in mind min heap will never increase and only decrease.  Min heap is a value that represents the low water mark of heap availability and really needs to be read in the context of the free and max buffer heap.  The max heap is the largest contiguous block of available memory while the free heap represents all blocks.  A small max buffer indicates memory fragmentation.  There will always be some because there is no garbage collector/compactor for heap on the ESP32 but when this memory gets small it will no longer be able to handle allocate the buffers for networking.  If you are not running the official release of 2.4.2 please make sure you update.  UDP responses were leaking previously.

> Why the board loses WiFi from time to time is another concern that I need to further investigate.

Check your router logs.  I do not see devices dropping off the network.   On occasion I do see the mDNS from the router not working properly if I don't have a DHCP reservation set up with the proper hostname in my router.   There is a bit of timing dance that goes on with it and routers do cute things with the .local or .lan domains.  Lots of wth on the interwebs about mDNS on ESP32.  If you are connecting externally the IP address is the most reliable.

alka79, 2024-04-24:
Thanks.
I am running 2.4.2 release.
Since I rebooted the board yesterday, no more Timing Somfy messages at idle anymore. There was something weird yesterday.

As per the heap, I realize my ES32-S2 has only 320MB RAM. That may explain a lot !

alka79, 2024-04-26:
I switched to an ESP32 board (Wemos Mini D2). No wifi connection losses anymore and min heap is now over 100K :)

ESP32-S2 with it's 320K RAM only seems to be too limited to properly run ESPSomfy.

closed 2024-04-26 by alka79