rstrouse/ESPSomfy-RTS #345: <Integration rstrouse/ESPSomfy-RTS> Repository structure for v2.4.2 is not compliant

GeoBau, 2024-04-24:
### Hardware

ESP32

### Firmware version

v2.4.2

### Application version

v2.4.2

### What happened? What did you expect to happen?

I cannot load the repository in HACS

### How to reproduce it (step by step)

```bash
1. goto Hacs
2. goto integrations
3. goto "benutzerdefinierte Repositories"
4. define "https://github.com/rstrouse/ESPSomfy-RTS" as repository
5. define "Integration" as category
6. Press add
7. Error: <Integration rstrouse/ESPSomfy-RTS> Repository structure for v2.4.2 is not compliant
```

### Logs

```bash
no logs
```

rstrouse, 2024-04-24:
You are trying to install the firmware repository into Home Assistant.  The proper url is https://rstrouse/ESPSomfy-RTS-HA.

GeoBau, 2024-04-25:
sorry - my mistake

closed 2024-04-25 by GeoBau