rstrouse/ESPSomfy-RTS #347: v2.4.2 position not working via http

wutz-ESP, 2024-04-25:
### Hardware

ESP32 WT32-ETH01

### Firmware version

v2.4.2

### Application version

v2.4.2

### What happened? What did you expect to happen?

v2.4.2 command http://192.168.10.89:8081/shadeCommand?shadeId=1&get=shades
["shadeId":1,"roomId":0,"name":"Dach WZ-rechts","remoteAddress":374797,"paired":true,"shadeType":3,"bitLength":56,"proto":0,"flags":1,"sunSensor":false,"hasLight":false,"repeats":1]

there are some things missing like position, downTime, upTime, lastRollingCode, ...

in v2.4.1 command http://192.168.10.89:8081/shadeCommand?shadeId=1&get=shades
gets direction | 0 | downTime | 49000 | flags | 1 | flipCommands | 0 | flipPosition | 0 | gpioDown | 0 | gpioUp | 0 | inGroup | 0 | lastRollingCode | 245 | light | 0 | myPos | -1 | myTiltPos | -1 | name | Dach WZ-rechts | paired | 1 | position | 0 | proto | 0 | remoteAddress | 374797 | repeats | 1 | roomId | 0 | shadeId | 1 | shadeType | 3 | sortOrder | 0 | stepSize | 100 | sunSensor | 0 | target | 0 | tiltDirection | 0 | tiltPosition | 0 | tiltTarget | 0 | tiltTime | 7000 | tiltType | 0 | upTime | 48000

Using fhem i get position asking http://192.168.10.89:8081/shadeCommand?shadeId=1&get=shades 

### How to reproduce it (step by step)

```bash
1. Go to...
2. Click on...
...
```

rstrouse, 2024-04-27:
In an effort to reduce some of the traffic on the response I shortened the return.   I will add the position data back in.

wutz-ESP, 2024-04-28:
### **this is a log file from installed firmware v2.4.1 MQTT traffic - v2.2.1 only displayed installed shades**

brackets at response turned from v2.4.1 { to v2.4.2 [

in v2.4.1 MQTT traffic is as much that fhem did not response anymore

2024-04-28_14:47:50 MQTT2_client_000000892815b80c status: online
2024-04-28_14:47:51 MQTT2_client_000000892815b80c ipAddress: 0.0.0.0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c host: ESPSomfyRTS-Buechner
2024-04-28_14:47:51 MQTT2_client_000000892815b80c firmware: v2.4.1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c serverId: 288900
2024-04-28_14:47:51 MQTT2_client_000000892815b80c mac: 0C:B8:15:28:89:03
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_shadeId: 1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_name: Dach WZ-rechts
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_remoteAddress: 374797
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_shadeType: 3
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_tiltType: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_flags: 1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_flipCommands: false
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_flipPosition: false
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_position: 100
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_direction: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_target: 100
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_lastRollingCode: 246
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_mypos: -1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_myTiltPos: -1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_1_windy: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_shadeId: 2
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_name: Dach EZ-links
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_remoteAddress: 374798
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_shadeType: 3
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_tiltType: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_flags: 1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_flipCommands: false
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_flipPosition: false
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_position: 100
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_direction: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_target: 100
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_lastRollingCode: 167
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_mypos: -1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_myTiltPos: -1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_2_windy: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_shadeId: 3
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_name: Markise WZ-rechts
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_remoteAddress: 374799
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_shadeType: 1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_tiltType: 2
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_flags: 1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_flipCommands: false
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_flipPosition: false
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_position: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_direction: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_target: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_lastRollingCode: 398
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_mypos: -1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_myTiltPos: -1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_tiltDirection: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_tiltPosition: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_tiltTarget: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_3_windy: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_4_shadeId: 4
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_4_name: Markise WZ-links
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_4_remoteAddress: 374800
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_4_shadeType: 1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_4_tiltType: 2
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_4_flags: 1
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_4_flipCommands: false
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_4_flipPosition: false
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_4_position: 0
2024-04-28_14:47:51 MQTT2_client_000000892815b80c shades_4_direction: 0
[... 785 lines cut ...]
2024-02-23_10:00:33 MQTT2_client_000000892815b80c shades_6_name: Markise EZ-links
2024-02-23_10:00:33 MQTT2_client_000000892815b80c shades_6_remoteAddress: 374802
2024-02-23_10:00:33 MQTT2_client_000000892815b80c shades_6_shadeType: 1
2024-02-23_10:00:33 MQTT2_client_000000892815b80c shades_6_tiltType: 2
2024-02-23_10:00:33 MQTT2_client_000000892815b80c shades_6_flags: 1
2024-02-23_10:00:33 MQTT2_client_000000892815b80c shades_6_flipCommands: false
2024-02-23_10:00:33 MQTT2_client_000000892815b80c shades_6_flipPosition: false
[... cut ...]
2024-02-23_10:00:33 MQTT2_client_000000892815b80c shades_6_windy: 0
2024-02-23_10:00:33 MQTT2_client_000000892815b80c 2: 2
2024-02-23_10:00:33 MQTT2_client_000000892815b80c 6: 6
2024-02-23_10:00:33 MQTT2_client_000000892815b80c 3: 3
2024-02-23_10:00:33 MQTT2_client_000000892815b80c 1: 1
2024-02-23_10:00:33 MQTT2_client_000000892815b80c 4: 4
2024-02-23_10:00:33 MQTT2_client_000000892815b80c 5: 5
2024-02-23_10:00:33 MQTT2_client_000000892815b80c groups: []
2024-02-26_10:51:11 MQTT2_client_000000892815b80c status: offline

rstrouse, 2024-04-28:
Actually what that log is reflecting is the dropping of unpublished topics.  Note the empty value.  It took me a second to think about what that log was indicating. That should have no impact on Fhem.

Also, if you are getting the shade data from the web services and not actually commanding the shade you can call http://ip address/shade?shadeId=xx

wutz-ESP, 2024-04-29:
thinking about having a module in fhem using websocket connection because this is getting every movement just in time.
actually i am using http://192.168.10.89:8081/shadeCommand?shadeId=1&get=shades 300 getting position every 5 minutes.

i am thinking about to much traffic asking all shades and blinds every 5 minutes if movements comes only 2 or 3 times a day.
If websocket information changes with v2.4.2 and gets less information it is better for me staying on older version.

what is the reason changing mqtt messages
from v2.2.1 where i get only shades / blinds configured in the web interface of
to v2.4.1 where i get messages from 32 shades and 
### this change is blocking my fhem mqtt2 server from 2024-04-29_06:34:24 to 2024-04-29_06:42:13

2024-04-29_06:34:24 MQTT2_client_000000892815b80c status: online
2024-04-29_06:34:24 MQTT2_client_000000892815b80c ipAddress: 0.0.0.0
2024-04-29_06:34:24 MQTT2_client_000000892815b80c host: ESPSomfyRTS-Buechner
2024-04-29_06:34:24 MQTT2_client_000000892815b80c firmware: v2.4.1
2024-04-29_06:34:24 MQTT2_client_000000892815b80c serverId: 288900
2024-04-29_06:34:25 MQTT2_client_000000892815b80c mac: 0C:B8:15:28:89:03
2024-04-29_06:34:25 MQTT2_client_000000892815b80c shades_1_shadeId: 1
2024-04-29_06:34:25 MQTT2_client_000000892815b80c shades_1_name: Dach WZ-rechts
2024-04-29_06:34:25 MQTT2_client_000000892815b80c shades_1_remoteAddress: 374797
2024-04-29_06:34:25 MQTT2_client_000000892815b80c shades_1_shadeType: 3
2024-04-29_06:34:25 MQTT2_client_000000892815b80c shades_1_tiltType: 0
2024-04-29_06:34:26 MQTT2_client_000000892815b80c shades_1_flags: 1
.
.
2024-04-29_06:34:40 MQTT2_client_000000892815b80c shades_6_tiltTarget: 0
2024-04-29_06:34:40 MQTT2_client_000000892815b80c shades_6_windy: 0
2024-04-29_06:34:40 MQTT2_client_000000892815b80c 1: 1
2024-04-29_06:34:40 MQTT2_client_000000892815b80c 6: 6
2024-04-29_06:34:40 MQTT2_client_000000892815b80c 5: 5
2024-04-29_06:34:40 MQTT2_client_000000892815b80c 3: 3
2024-04-29_06:34:40 MQTT2_client_000000892815b80c 4: 4
2024-04-29_06:34:40 MQTT2_client_000000892815b80c 2: 2
2024-04-29_06:34:40 MQTT2_client_000000892815b80c shades_7_shadeId: 
2024-04-29_06:34:41 MQTT2_client_000000892815b80c shades_7_name: 
2024-04-29_06:34:41 MQTT2_client_000000892815b80c shades_7_remoteAddress: 
.
.
2024-04-29_06:42:09 MQTT2_client_000000892815b80c shades_32_windy: 
2024-04-29_06:42:09 MQTT2_client_000000892815b80c shades_32_sunny: 
2024-04-29_06:42:09 MQTT2_client_000000892815b80c groups: []
2024-04-29_06:42:09 MQTT2_client_000000892815b80c groups_1_groupId: 
2024-04-29_06:42:09 MQTT2_client_000000892815b80c groups_1_name: 
2024-04-29_06:42:09 MQTT2_client_000000892815b80c groups_1_remoteAddress: 
.
.
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_15_SunSensor: 
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_15_flipCommands: 
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_16_groupId: 
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_16_name: 
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_16_remoteAddress: 
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_16_groupType: 
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_16_direction: 
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_16_lastRollingCode: 
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_16_flags: 
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_16_SunSensor: 
2024-04-29_06:42:13 MQTT2_client_000000892815b80c groups_16_flipCommands:

rstrouse, 2024-04-29:
Actually the shadeCommand endpoint you are using is not the correct endpoint to get the shade state.  That endpoint is used to initiate movement.  The reason for the reduction in response payload is to process these commands as quickly as possible.

If you want to poll it is more efficient to get a single payload using

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Integrations#get-shades

While rest commands using polling are great the connection is opened and closed between each call and has to be renegotiated on each request.  Sockets maintain this connection and operate as a just in time event.

rstrouse, 2024-05-08:
I suppose I should have looked closer at your original documentation.  I did not notice that the return was actually an array instead of an object.   This has been fixed in the pre-release v2.4.3

rstrouse, 2024-05-12:
I did fix the improper return from the shadeCommand endpoint.  I am happy to help if you need anything on getting the FHEM to work with the sockets.   I am going to close this for now but let me know if you need help.

closed 2024-05-12 by rstrouse

wutz-ESP, 2024-05-13:
Sorry, at the moment i have to do other things and no time.First thing to do is creating another testing device.Thank you offering help getting fhem working with this great device.Ist this contact ok for asking help later this year?Kind regards--Diese Nachricht wurde von meinem Android Mobiltelefon mit 1&1 Mail gesendet.